TechxArtisanStudio/Openterface_QT #3: Clarify code license

justinas, 2024-05-24:
Excited to see the first version of the code! I noticed that `main.cpp` includes the following, indicating dual-licensing via either Qt commerical license, or BSD:

https://github.com/TechxArtisanStudio/Openterface_QT/blob/56f32105f9520d1a874450770901d460f2321da3/main.cpp#L2

However the `LICENSE` file states AGPL 3. Presumably the license of `main.cpp` is because it was based upon Qt sample code?

In any case, please do clarify in regards to license, is it AGPL 3 (v3 only, or "v3 or later"?).

kevinzjpeng, 2024-05-25:
Thanks for pointing out the license inconsistency. You are right the " Qt commerical license, or BSD" license shows in main.cpp came from the QT sample programe. I already corrected all the files which have wrong license header.

kevinzjpeng, 2024-05-25:
This project is principally licensed under the Affero General Public License (AGPL) version 3. At this time, we have chosen to adhere to the 'v3 only' terms to ensure consistency and clarity regarding the sharing and modification of our project.

However, we understand the importance of adaptation and evolution in the open-source world. Therefore, while we are firm on the AGPLv3 at present, we remain open to considering later versions if future circumstances deem it beneficial and necessary for the continued growth and openness of our project.

kevinzjpeng, 2024-05-29:
Really thanks for spotting out the header issue! To be honest, it's copy from another opensource project and I overlooked the detail, I have fixed now.

closed 2024-05-29 by kevinzjpeng