TechxArtisanStudio/Openterface_QT #5: Feature request: Target CD-ROM / USB storage emulation

Trapsplz, 2024-06-05:
From the feature list of the product I strongly miss the opportunity to emulate target CD-ROM station and/or USB storage devices. This would be very useful for directly mounting ISO files or directories directly from the Host computer (i.e for installing operator system etc).

RyanWor, 2024-06-05:
This would be a great idea and other tools have accomplished (things like the isostick, or Dell iDRAC/HP iLow). I am not certain if that is something the team could implement purely with software though. Definitely would be cool if it was possible. 

An easier approach might be to create a bootable USB or use something like an isostick (pretty sure support for them died off sadly) and connect that to the shared USB port. You could add ISO, change the boot order/selection, etc. while switched to Host and then switch it to Target when ready to boot with it.

Trapsplz, 2024-06-05:
Opensource KVM-over-IP solutions such as [pikvm](https://github.com/pikvm/kvmd/tree/master/kvmd/apps/otgmsd), [blikvm ](https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-msd/kvmd-msd.sh)and [piLO](https://zach.bloomqu.ist/blog/2020/08/pilo-raspberry-pi-lights-out-management.html) have similar features, Its a bit "beyond my level" but i'm sure your team will see quickly whether it is feasible with the current design or not :) 

Since the use-case here is mostly for people within arms reach of the target hardware one could probably argue if its worth implementing such a feature thou, since (as you point out) its easy to keep an iso-loaded stick on hand. Check out [Ventoy ](https://www.ventoy.net/en/index.html) if you havent already btw, its an opensource project very similar to the isostick you mention :) But it would be a neat feature and worth a consideration i think ;)

RyanWor, 2024-06-06:
FYI I am not on the openterface team, just another beta tester. I totally hope something like this could be integrated, I am just not sure the hardware is in the openterface devices to do it the way PiKVM, BliKVM, and others can. 

I still have my isostick actually but I don't think it works well on most modern machines. I will check out Ventoy though, thank you!

kevinzjpeng, 2024-06-06:
Thank you, Trapsplz, for your feature request and the useful information shared. Also, I appreciate Ryan's insightful input.

Regarding the possibility of emulating CD-ROM or USB storage like PiKVM, BliKVM or piLO, I'm afraid the current architecture of Openterface doesn't support this feature. This is primarily due to two reasons:

1. The decision to keep power usage minimal for stability purposes and to allow the device to be powered via a USB 2.0 cable. As a result, we had to opt for a functional MCU over a Linux-based architecture, which would have demanded more power.

2. We aimed to offer users the flexibility to use any storage device of their liking and to easily switch storage between host and target. This led to the inclusion of a toggle button function.

Given these design considerations, Openterface does not have an underlying OS, and the chipset currently in use, unfortunately, does not support OTG USB port emulation.

I came across Ventoy shared by Trapsplz and found it to be an excellent solution. Ventoy, combined with Openterface, should sufficiently address the issues you are encountering.

Trapsplz, 2024-06-06:
Thanks for your reply Kevin.
Maybe a feature to consider for the next product or iteration from openterface team then :)

I will mark the issue as "closed, not planned".

Keep up the good work! And the best of luck with product release, must be very exciting times for the team!

closed 2024-06-06 by Trapsplz

RyanWor, 2024-06-06:
@Trapsplz This was posted in discord last night by Kevin, and might be handy when it comes to using something like a ventoy USB drive with an openterface:

"Glad to update everyone a minor change on the hardware, we can use software to control the host/target toggle button, which means if we enable VNC or MJPEG streaming in future, you can enjoy to remote usb device toggle. Hopefully we can complete all testing before production. 🤞"

Trapsplz, 2024-06-06:
Neat! By the way, is the discord open to the public? Or only testers?

RyanWor, 2024-06-06:
It's public: https://openterface.com/community/

The direct link to Discord is: https://discord.gg/sFTJD6a3R8

Trapsplz, 2024-06-07:
Thanks! I'll join in when I get off work :D

kashalls, 2024-06-13:
You should be able to use a USB with Ventoy with the Openterface with "Target" mode on. I'm planning on testing this later this week.

Trapsplz, 2024-06-13:
Sounds promising! Let us know if it works out :D