TechxArtisanStudio/Openterface_QT #9: Use the host keyboard layout instead of the target's

shadowkora, 2024-06-26:
Current implementation only reflects the keyboard of the target, so if the host machine has QWERTZ layout and the target has QWERTY, it will be typing as QWERTY, so I think some translation layer is needed to reflect the host machine's layout on the target machine.

In my example, I also use the CES QWERTZ layout, when typing on an Ubuntu machine that uses ENG QWERTY, numbers on my the top row are not getting passed to the host machine at all, typing 1 2 3 4 results in = - - - (- represents nothing being typed on target)

d3vyce, 2024-07-07:
Same issue for me with AZERTY (host) -> QWERTY (target).

CableCatDK, 2024-12-11:
I have a Danish leyboard at both the host and target PC. But a lot of keys are not passed through, and some are wrong. It would be great with an option to pass the scan-codes dicrectly through. It would also be great to specify the keybord on both the host and target PC. So you could have a mix of 104 and 105 key keyboards, and languages.

glennbossitbe, 2025-01-31:
Same issue for me with AZERTY - Belgian point (host) -> QWERTY (target).
Any update on this?

CableCatDK, 2025-01-31:
I decided to look into this issue myself. Since the same issue is in VMware, and never got fixed.

**I have succesfully build the source:**
Download [Qt Online Installer for Windows (x64)](https://d13lb3tujbc8s0.cloudfront.net/onlineinstallers/qt-online-installer-windows-x64-4.8.1.exe)
You do not need to install everything. Just
- QT creator
- Qt 6.5.3
- QT Serial Port
- Qt Multimedia

**Analyse of source**
The file [target/Keymapping.h](https://github.com/TechxArtisanStudio/Openterface_QT/blob/ac610380b58869590d4f9d4c94f56d8ab424c99e/target/Keymapping.h#L137) has the mapping between what the host OS recieve, and the scancode to be send to the target.
The file [target/KeyboardManager.cpp](https://github.com/TechxArtisanStudio/Openterface_QT/blob/ac610380b58869590d4f9d4c94f56d8ab424c99e/target/KeyboardManager.cpp#L46) select which host keyboard map is used

There are two keyboard maps:
- Default (US)
- UK, selected if locale is set to UK (not UK keybord)

**What I suggest is this**
- There no need for the user to input the host keyboard. This is already done to the OS.
- There must be a way to choose target keyboard. The default could be to match the host OS keyboard.
- There must be a database of keymapping that can be reused for this project.

CableCatDK, 2025-02-01:
I have a quick hack/workaround: set the host keyboard to US, and the target keyboard to your actually keyboard layout.

This will pass the keys correctly through. Expect the alt-gr does not work.

I have looked more into Qt: It does support reading the host keyboard scan code. Which are the same regardless is the selected keyboard layout.

These keys can be translated from PS/2 to HID and send to the host. Making an universal solution.