TechxArtisanStudio/Openterface_QT #41: Tab & Numberpad not funcitoning as intended in lastest release.

kashalls, 2024-09-02:


kevinzjpeng, 2024-09-03:
Thanks for reporting this, fixed one of the issue reported, but the other cannot be reproduced.

- Tab results in serial error, causing a disconnect until Qt is closed and reopened. 
(FIXED) This is because QT recognized TAB key as change focus in the windows, I override the default focusNextPrevChild method to fix it, see this commit: https://github.com/TechxArtisanStudio/Openterface_QT/commit/a08debd8a5699bc234d9e6d2a5bbaa586d00442d.

- Num Lock is enabled, yet 0 returns as a Insert key (**Cannot Reproduced**)
- 1 is end, 2 is down arrow, 3 is page dn etc.... not typing numbers.(**Cannot Reproduced**)

Can you share what's the keyboard layout and more keyboard setting? I guess some config difference between us.

kashalls, 2024-09-03:
> Can you share what's the keyboard layout and more keyboard setting? I guess some config difference between us.

It's a QWERTY layout. Should just be a standard US keyboard with keypad attached. It didn't matter if NumLock is enabled or disabled, it still didn't use the right keys. I wonder if this is host issue.

I just did the same keyboard tester on both host and target, Qt doesn't update the num lock correctly?

kevinzjpeng, 2024-09-06:
Can you enable Host logging from File-> Perference -> General, and use command line to start openterface, then click the button with NumLock on and off, then update the log to this issue. Thanks.
(We should import debug logging to more user friendly, LOL)

Chow411, 2025-06-11:
Lastest version had fix this issue

closed 2025-06-11 by Chow411