TechxArtisanStudio/Openterface_QT #112: Input to target does not work on Windows 11 host

CableCatDK, 2025-01-08:
I am using the Mini KVM in Widnows 11. It is my exprince that most of the time keyboard and mouse does not work on the target. Some rare times it suddenly works.

I have this issue on two differt Widnows 11 PCs with two differt Mini KVMs.

Is it just me? or is this a general problem?

kevinzjpeng, 2025-01-08:
Did you brought two single KVM version or toolkits version? 

If yours are two single KVM version, please make sure you use high quality usb cables. The cables really matters on the keyboard and mouse control stability.

If yours are the toolkit version, can you please share your wiring setup?

CableCatDK, 2025-01-09:
- It is the Mini-KVM Toolkit I bougth.
- I am using the high quality UBS-C cable that came with the kit (not the organge one).

I have tried a few differnt cable setups. Right now the KVM is working with this setup:

Target: RPI4
Host: Windows 11
Video: RPI Micro HDMI output > Raspberry Micro HDMI to HDMI converter > Openterface HDMI cable > MiniKVM input
HID: MiniKVM USB-C target > IKEA USB-C to USB-A cable > RPI USB
Host <> PC: MiniKVM USB-C host <> Openterface USB-C 20Gbs cable <> ThinkVision P40w-20 <> USB2.0 10 meter active cable <> PC

[image]

Is there something I can do next time it stops working, to figure out what is the issue?

CableCatDK, 2025-01-09:
I was asked about with software I was using:
The download says verison 0.0.3, but the about page says 0.0.2.
[image]

kevinzjpeng, 2025-01-10:
> * It is the Mini-KVM Toolkit I bougth.
> * I am using the high quality UBS-C cable that came with the kit (not the organge one).
> 
> I have tried a few differnt cable setups. Right now the KVM is working with this setup:
> 
> Target: RPI4 Host: Windows 11 Video: RPI Micro HDMI output > Raspberry Micro HDMI to HDMI converter > Openterface HDMI cable > MiniKVM input HID: MiniKVM USB-C target > IKEA USB-C to USB-A cable > RPI USB Host <> PC: MiniKVM USB-C host <> Openterface USB-C 20Gbs cable <> ThinkVision P40w-20 <> USB2.0 10 meter active cable <> PC
> 
> ![IMG_0131](https://private-user-images.githubusercontent.com/13363063/401452562-aaaa6595-502e-42b9-a1c3-b8b0c20b916e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY0NzU0MTIsIm5iZiI6MTczNjQ3NTExMiwicGF0aCI6Ii8xMzM2MzA2My80MDE0NTI1NjItYWFhYTY1OTUtNTAyZS00MmI5LWExYzMtYjhiMGMyMGI5MTZlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDAyMTE1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFkNTY3NmNhMTQ1MjM1MmExNDE2ZmU0YzIzMzIxM2M2YTM4Y2EzMzZjZWVkZDU4MmVjZmYzMjAwMWI1NDY1MzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.E8TD_arjdcPUG6r_olj61yTB5Zl9YYKLgG0E9ZCE67k)
> 
> Is there something I can do next time it stops working, to figure out what is the issue?

It's good to hear that you have a working setup now. However, I noticed that you are using an "IKEA USB-C to USB-A cable" instead of the "Type C to USB-A cable" provided in the Openterface toolkit. I believe that the cable included in the Openterface package will be more stable due to its shorter length.

The Raspberry Pi 4 (or RPI5) is known to have power supply issues due to its high power requirements. Using a low-quality power supply can result in unstable USB port voltage on the Pi, which can affect the stability of the Openterface HID chip. Both the target and host computers supply power to the Openterface, and typically, the shorter cable acts as the main power source. Therefore, I suggest using the default Openterface toolkit "Type C to USB-A cable" for better stability.

If the KVM stops working again, consider the following steps to troubleshoot and identify the issue:

1. Check Power Supply.
2. Inspect Cable Connections
3. Switch Cables.
4. Alternative Ports, usually USB 3.0 ports have better power supply (max 900mA output, but 500mA only on 2.0)
5. If possible, check the host and target side USB ouput voltage by USB Multimeter.

kevinzjpeng, 2025-01-10:
> I was asked about with software I was using: The download says verison 0.0.3, but the about page says 0.0.2. [...]

From your picture, you are using quite old version. And I found the offical website is still pointing to an old version. Sorry, I will fix it. Please try this newer software in this link.

https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/v0.0.6

CableCatDK, 2025-01-10:
Hi here is an update:
Yesterday I left the MiniKVM Window app open, in a working state. After around 4 hours I swiched back to it, and the input was no longer working. The issue was not fixed by repluging the MiniKVM or restarting the app. I turned the PC off at night. When I stated the PC this morning, the MiniKVM was working agian.

The RPI4 is powered by its orginal PSU from Raspberry Pi.

I will update the app, and use the C to A USB cable provided in the toolkit.

CableCatDK, 2025-01-12:
The [v0.0.6](https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/v0.0.6) seems more stable. But I have a reurcurring issue: When the setup have been idel for some time. Then the keyboard and mouse no longer works. No input is registered in the console. There 3 ways workaround the issue:

- Repluing the host USB-C port (hard reset)
- Repluging the target USB-C port
- Restart the serial port from the gui.

Video of issue: https://youtu.be/vh92a3fBTCM

CableCatDK, 2025-01-17:
You sould update the link at https://openterface.com/app/ It is still v0.0.3

kevinzjpeng, 2025-01-30:
Thanks for reminding the version issue in the web page, I have updated them all with dynamic build script.