TechxArtisanStudio/Openterface_QT #116: Wrong version in release

DanielWeigl, 2025-01-16:
Downloaded the newest version 0.0.6 here https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/v0.0.6

```
# wget https://github.com/TechxArtisanStudio/Openterface_QT/releases/download/v0.0.6/openterfaceQT.linux.amd64.deb
...

# sha256sum openterfaceQT.linux.amd64.deb 
af626220ede1752a42fdf962cd253e41dfd61da565d82bcdcba9e6858333196a  openterfaceQT.linux.amd64.deb

# sudo dpkg -i openterfaceQT.linux.amd64.deb     
(Reading database ... 492570 files and directories currently installed.)
Preparing to unpack openterfaceQT.linux.amd64.deb ...
Unpacking openterfaceqt (0.0.39.009) over (0.0.39.009) ...
Setting up openterfaceqt (0.0.39.009) ...
```

but this installs `0.0.39.009` instead of `0.0.6` - also the app shows the 0.0.39 version in the title bar.

[image]

nicojako, 2025-01-31:
I have the same issue

hawk6242, 2025-02-02:
I also see the same issue on the windows release

kevinzjpeng, 2025-02-11:
It was fixed in https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/v0.1.0, so I close this issue.