TechxArtisanStudio/Openterface_QT #118: Installation fails on Mint 21.3.

BertQuodge, 2025-01-29:
I use Linux Mint 21.3 with Cinnamon and followed the installation instructions for Linux. On typing "openterfaceQT" I receive this error:

openterfaceQT: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by openterfaceQT)
openterfaceQT: /lib/x86_64-linux-gnu/libQt6Core.so.6: version `Qt_6.4' not found (required by openterfaceQT)

Thanks

kevinzjpeng, 2025-01-30:
The error messages suggest there might be missing or outdated dependencies (`GLIBCXX_3.4.32` and `Qt_6.4`). I’ve tested Openterface_QT on Linux Mint 21.3 and didn’t encounter this problem, so let’s troubleshoot a few things:  

1. **Dependency Check:**  
   Could you double-check that you’ve installed all the required dependencies listed in the [Linux installation guide](https://github.com/TechxArtisanStudio/Openterface_QT?tab=readme-ov-file#for-linux-users)? Specifically:  
   ```bash
   sudo apt install -y libqt6core6 libqt6dbus6 libqt6gui6 libqt6network6 libqt6multimedia6 libqt6multimediawidgets6 libqt6serialport6 libqt6svg6 libusb-1.0-0-dev
   ```  
   If any of these commands threw an error during installation, please let me know.  

2. **GLIBCXX Update:**  
   For the `GLIBCXX_3.4.32` error, try updating `libstdc++6`:  
   ```bash
   sudo apt install -y libstdc++6
   ```  
   If this doesn’t work, you may need to upgrade to a newer GCC version (e.g., via `ppa:ubuntu-toolchain-r/test`).  
   If it works, then you should revist the step 1 again.

3. **Qt6 Version:**  
   The error `Qt_6.4 not found` indicates your system might not have Qt6.4 installed. Linux Mint 21.3’s default repositories may include an older Qt version. To resolve this:  
   - Confirm your Qt6 version with:  
     ```bash
     qmake6 --version
     ```  
   - If it’s not Qt6.4 or newer, you may need to install Qt6.4 or newer manually, you can check this [link](https://doc.qt.io/qt-6/linux.html#building-qt-from-source), it's quite time consuming.

Let me know the output of these steps, and I’ll help you troubleshoot further!

glandecki, 2025-01-30:
Great tool that works only with a 'test' versions of OSes.. All stable versions are using gcc12 like RHEL9/debian 12 etc.
Definitely I will install the test version on my laptop to use it.. and crash all other things..

eh..
Another 'great' tool for hipsters, but not professionals..

randrews, 2025-01-30:
I have this same problem, but I'm running Debian 12. What version of gcc is required, so I can find a Debian PPA to install it from?

(any chance of releasing a static binary?)

glandecki, 2025-01-30:
You have to build it from source per instructions.. no changes required, but again shame not to support the stable versions

randrews, 2025-01-30:
You are correct! Building it with gcc 12 worked fine. It does _support_ the stable versions, just the posted binary wasn't built with the stable version. I assumed it would require 14 to build, which it doesn't.

BertQuodge, 2025-01-30:
Thanks for your help @kevinzjpeng, my QT version is the problem and isn't something I can easily resolve on multiple machines.

I wonder if we need a flatpak, or similar? I currently utilise a PiKVM in my grab bag, meaning I can get console access via any laptop I can use. I set the PiKVM as a wifi AP, so I just wifi from a laptop with a browser to gain console access. For me, to replace the PiKVM, I need to be able to use any laptop to hand, which could be running Windows or Linux. I don't mind carrying a USB memory stick with the files I might need. Saying that many corporate laptops block USB ports, but YMMV.

Perhaps my use case isn't the norm, I'm just sharing what I would need.

Thanks

glandecki, 2025-01-30:
I think it's OK when you can easily compile it, but it just is a bit of a hassle for some users (as I assume the software will develop and you'll have to do it again and again). It would be great to have a package built for stable versions of OS-es not 'testing' ones. This tool will be used for many sysops/devops or just other good-old-time admins that are not running the test OSes on their systems.

BertQuodge, 2025-01-31:
I just spent time manually building. As @glandecki says, this isn't a long term viable option (for me). Perhaps many users will only use the interface on a computer or 2, not in the enterprise. My use case is to use the openterface on any PC I have to hand as a crash cart solution, I can't be compiling code when I need to get console access to a device at short notice. The interface needs to work quickly and first time, every time, for any PC I can get my hands on.

I'll continue with my PiKVM but I'll return in a few months to see how the openterface project is going. I see potential with the product, it might just need time to develop. The openterface unit is built well, and is more elegant than my home built PiKVM with handmade USB OTG cable.

kevinzjpeng, 2025-02-02:
I am working on other issues fix and it's still in CNY holiday now, sorry for repling late, and appreacate for the discussion here to make it clear.

> I have this same problem, but I'm running Debian 12. What version of gcc is required, so I can find a Debian PPA to install it from?
> 
> (any chance of releasing a static binary?)

@randrews, Regarding to the GCC version, I have downgrade the Github action runner, and fixed in this build, you can test it if you cannot wait for my next release. 

The CI build for the fix, https://github.com/TechxArtisanStudio/Openterface_QT/actions/runs/13088097594
FYI, for another discussion here, https://github.com/TechxArtisanStudio/Openterface_QT/issues/129

Regarding to the static binary, I will take it and put into my backlog,  thanks for the idea.

kevinzjpeng, 2025-02-02:
> Thanks for your help [@kevinzjpeng](https://github.com/kevinzjpeng), my QT version is the problem and isn't something I can easily resolve on multiple machines.
>
> I wonder if we need a flatpak, or [...]

Thank you for sharing your detailed use case! While Openterface Mini-KVM currently relies on Qt for its cross-platform GUI, we fully recognize the challenges posed by corporate restrictions (like blocked USB ports) and dependency management across machines. 

Yes, we will distribute the app to flatpak or distrobox, this is in our plan, please stay tuned. By the way, will your company block flatpak or distrobox on a linux as well?