TechxArtisanStudio/Openterface_QT #120: Double click unreliable

randrews, 2025-01-31:
Most of the time, double clicking the mouse doesn't get sent to the target, or it's sent as one click. I can see two click events being sent in the console, but they're not being sent fast enough, or too fast, or something, because the target OS doesn't register it as a double click.

```[2025-01-30 18:29:09][140124004313472] {Debug}: "Data written to serial port: @ttyUSB0" : "57 ab 00 04 07 02 00 1c 09 a4 04 00 dc"
[2025-01-30 18:29:09][140124004313472] {Debug}: Receive from serial port @ 115200 : "57 ab 00 84 01 00 87"
[2025-01-30 18:29:09][140124004313472] {Debug}: Absolute mouse event sent, status 
[2025-01-30 18:29:10][140124004313472] {Debug}: "Data written to serial port: @ttyUSB0" : "57 ab 00 04 07 02 00 1c 09 a4 04 00 dc"
[2025-01-30 18:29:10][140124004313472] {Debug}: Receive from serial port @ 115200 : "57 ab 00 84 01 00 87"
[2025-01-30 18:29:10][140124004313472] {Debug}: Absolute mouse event sent, status 
```

I've turned the double click speed on the target OS down as far as it'll go, doesn't seem to affect it. A mouse plugged directly into the target machine can double click no problem.

Host OS Debian 12, target OS Windows 11.

randrews, 2025-01-31:
Here's a PR that fixes it: https://github.com/TechxArtisanStudio/Openterface_QT/pull/122

kevinzjpeng, 2025-02-03:
Thanks for the PR, I have merged and suppose it's working well in Debian 12, I will close this issue.

closed 2025-02-03 by kevinzjpeng