TechxArtisanStudio/Openterface_QT #121: 4:3 video modes, cursor position is scaled incorrectly

randrews, 2025-01-31:
If the target machine is set to 1024x768 resolution, and the mouse to absolute mode, the mouse cursor coordinates get translated wrong: the window has black bars on the sides, and these black bars count toward the absolute position, so if the cursor is 25% from the left edge of the window (including black bar), the cursor in the target machine will be 25% from the left edge of its screen (which has no black bar). So the cursor will always be off to the side, unless it's in the exact center. See attached video:

https://github.com/user-attachments/assets/f653a40e-66b8-49fc-ad9f-c4f05afcf197

randrews, 2025-01-31:
Easy fix for this would be to not force the window into 16:9 aspect ratio, if it matched the ratio of the video signal then it wouldn't have black bars, no scaling issue.

zorxd, 2025-02-27:
Looks similar to https://github.com/TechxArtisanStudio/Openterface_QT/issues/136

randrews, 2025-02-27:
> Looks similar to [#136](https://github.com/TechxArtisanStudio/Openterface_QT/issues/136)

The black bar part can't be fixed because hardware limitation, but the mouse offset part can be, that's Qt parsing the mouse position wrongly.