TechxArtisanStudio/Openterface_QT #125: Ubuntu 22.04 deb installation fails for 0.0.6

Fescron, 2025-01-31:
Hello, I just received my mini-KVM and wanted to try it out on Ubuntu 22.04.5 LTS. In the readme it says that 22.04 is supported, but I can't seem to install `openterfaceQT.linux.amd64.deb`.

I have tried all of the steps before actualling installing the above package, but it already fails at the command installing the dependencies:
```bash
The following packages have unmet dependencies:
 openterfaceqt : Depends: libqt6core6 (>= 6.4.2) but 6.2.4+dfsg-2ubuntu1.1 is to be installed
                 Depends: libqt6multimedia6 (>= 6.4.2) but 6.2.4-1 is to be installed
                 Depends: libqt6multimediawidgets6 (>= 6.4.2) but 6.2.4-1 is to be installed
                 Depends: libqt6serialport6 (>= 6.4.2) but 6.2.4-1 is to be installed
                 Depends: libqt6svg6 (>= 6.4.2) but 6.2.4-1ubuntu1 is to be installed
```

This must be because my qt version is not high enough?
```bash
qmake6 --version
      QMake version 3.1
      Using Qt version 6.2.4 in /usr/lib/x86_64-linux-gnu
```

I also tried building from source, but then `qt6-multimedia-dev` and `qt6-serialport-dev` can't be found...

I haven't fully looked into trying to somehow install a newer QT version, I suppose I need to do that?

Thank you!

kevinzjpeng, 2025-02-01:
Thank you for reporting the issue! The issue you’re encountering is indeed due to the Qt6 version on Ubuntu 22.04 being older than what OpenTerfaceQT requires. Ubuntu 22.04 ships with Qt 6.2 by default, but OpenTerfaceQT needs Qt 6.4 or newer. Below are steps to resolve this:  

---

### **Option 1: Using Debian Bookworm Repositories (Please be aware for the Risk)**  
Although this Option was tested by me, adding Debian repositories which will  provide newer Qt6.4 packages, 
**this is discouraged** unless you fully understand the risks. Following are the potential risks I know about.

**Risks:**  

1. Dependency Hell - Debian’s Qt6.4 may depend on newer versions of libraries (e.g., libstdc++, gcc) that conflict with Ubuntu’s older versions. This can cause unresolvable dependency errors.

2. Security Vulnerabilities - Mixing repositories bypasses Ubuntu’s security updates. Debian’s packages might not receive timely backported fixes for Ubuntu-specific vulnerabilities.

3. Unsupported Configuration - Canonical does not support hybrid Debian/Ubuntu systems. Future updates (e.g., dist-upgrade) may fail catastrophically.

**Steps:**  
1. Backup your system or test in a virtual machine first.  
2. Edit `/etc/apt/sources.list` and add:  
   ```bash
   deb http://deb.debian.org/debian bookworm main contrib non-free-firmware  
   deb http://deb.debian.org/debian bookworm-updates main contrib non-free-firmware  
   deb http://deb.debian.org/debian-security bookworm-security main contrib non-free-firmware  
   ```  
3. Set **priority pinning** to avoid overriding Ubuntu packages:  
   Create `/etc/apt/preferences.d/99-debian-bookworm` with:  
   ```  
   Package: *  
   Pin: release n=bookworm  
   Pin-Priority: 50  
   ```  
   (This ensures Debian packages are only installed explicitly, not automatically.)  
4. Update and install Qt6.4 packages:  
   ```bash  
   sudo apt update  
   sudo apt install -t bookworm libqt6core6 libqt6dbus6 libqt6gui6 libqt6network6 libqt6multimedia6 libqt6multimediawidgets6 libqt6serialport6 libqt6svg6 libusb-1.0-0-dev
   ```  
 5. Install openTerfaceQT again.  
 6. If openTerfaceQT works, rollback the step 2 & 3 to avoid protential risk in future.

---

### **Option 2: Manual install Qt6 from installer**  
If the above options fail, use the **Qt Online Installer** to install Qt6.4+ manually, recommand 6.5.4:  
1. Download the source installer from [Qt’s website](https://www.qt.io/download-qt-installer-oss).  
2. Run the installer and select Qt 6.4+ under “Open Source” licensing.  
3. Ensure the Qt6 paths are added to your environment (e.g., `PATH`, `LD_LIBRARY_PATH`).  

---

### **Building from Source**  
After upgrading Qt, you can run the openterfaceQT.linux.amd64.deb again to see if it works.

---

On the other head, I am working on a script and a docker environment to make the app easy to run in different Linux. Thanks for your patient!

Fescron, 2025-02-01:
Thank you for the fast and thorough response! Hmm, that seems quite a bit of work and could indeed cause instability. The script you are talking about, will that be used to build a .deb for 22.04 with QT6.4 then? Or will OpenTerfaceQT never be able to support that version? In the meantime I suppose I can dual-boot back into Windows and try it there :)

kevinzjpeng, 2025-02-01:
The script is done, it help you to buld the QT and dependency. When I did my testing on Ubuntu 22.04, I acutally used the option 1 to get the QT 6.4 environment from Debain repository.

You should save it to a folder and run with sudo.  The script will auto build and install the official QT 6.5.3 environment and FFMPEG 6.6.1 from source for Ubuntu 22.04, it tested in my Ubuntu 22.04, it works well but take around 20 minutes to complete.

If everything goes well, you can run the openterfaceQT.linux.amd64.deb again.

Here is url of the script:
https://github.com/TechxArtisanStudio/Openterface_QT/blob/main/docker/install-qt-env.sh

Please let me know if it helps.

Fescron, 2025-02-01:
Hmm, I tried it on a "test laptop" with also Ubuntu 22.04 and at first I had to also install `yasm`. Everything was going well until I got the following error:

```
...
FAILED: src/plugins/multimedia/ffmpeg/CMakeFiles/QFFmpegMediaPlugin.dir/qx11surfacecapture.cpp.o
...
/home/brecht/Downloads/qt-build/qtmultimedia/src/plugins/multimedia/ffmpeg/qx11surfacecapture.cpp:21:10: fatal error: X11/extensions/Xrandr.h: No such file or directory
   21 | #include <X11/extensions/Xrandr.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
...
[270/271] Building CXX object src/plug...FmpegMediaPlugin.dir/qv4l2camera.cpp.o
ninja: build stopped: subcommand failed.
```

Euhm, is this because my Ubuntu installation uses Wayland?

kevinzjpeng, 2025-02-02:
Your Ubuntu may miss the libxrandr-dev package, you can try following command to get it.

```bash
sudo apt install libxrandr-dev
```
Let me know if it works and I will update the script.

Fescron, 2025-02-02:
Yes by installing that package I was able to execute the script from beginning to end.

I think we are getting close, but there is still something missing. The installed QT files are in the PATH:
```
echo $PATH
   /opt/QT6/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
```

However, when I again do `sudo dkpg -i openterfaceQT.linux.amd64.deb` I get the following error (which is the same as before I think?)

```
Selecting previously unselected package openterfaceqt.
(Reading database ... 275785 files and directories currently installed.)
Preparing to unpack openterfaceQT.linux.amd64.deb ...
Unpacking openterfaceqt (0.0.39.009) ...
dpkg: dependency problems prevent configuration of openterfaceqt:
 openterfaceqt depends on libqt6core6 (>= 6.4.2); however:
  Version of libqt6core6:amd64 on system is 6.2.4+dfsg-2ubuntu1.1.
 openterfaceqt depends on libqt6multimedia6 (>= 6.4.2); however:
  Version of libqt6multimedia6:amd64 on system is 6.2.4-1.
 openterfaceqt depends on libqt6multimediawidgets6 (>= 6.4.2); however:
  Version of libqt6multimediawidgets6:amd64 on system is 6.2.4-1.
 openterfaceqt depends on libqt6serialport6 (>= 6.4.2); however:
  Version of libqt6serialport6:amd64 on system is 6.2.4-1.
 openterfaceqt depends on libqt6svg6 (>= 6.4.2); however:
  Version of libqt6svg6:amd64 on system is 6.2.4-1ubuntu1.

dpkg: error processing package openterfaceqt (--install):
 dependency problems - leaving unconfigured
Processing triggers for mailcap (3.70+nmu1ubuntu1) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu3) ...
Processing triggers for desktop-file-utils (0.26-1ubuntu3) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Errors were encountered while processing:
 openterfaceqt
```

In your first reply you mentioned `LD_LIBRARY_PATH`, do I also need to do something with that?

Thank you for the quick responses and support!

Fescron, 2025-02-03:
Getting closer again! It now basically works (I also had to remove `brltty` to get mouse input working) except that there seems to be an incorrect font used...

[image]

Fescron, 2025-02-03:
(Also, if I close the application the last line in the terminal says `Segmentation fault (core dumped)`. The rest of the debug information above this line seems to be okay)

kevinzjpeng, 2025-02-11:
> Getting closer again! It now basically works (I also had to remove `brltty` to get mouse input working) except that there seems to be an incorrect font used...
> 
> [image]

Yes, it's getting very close, the font cannot be shown because your system is lack of opensource font.

You can try to install them manually and build again.

``` bash
sudo apt install -y libfontconfig1-dev libfreetype6-dev
```
Or you can try the following command for one line build from source.

``` bash
curl -sSL https://raw.githubusercontent.com/TechxArtisanStudio/Openterface_QT/refs/heads/main/docker/build-from-source.sh | sudo bash
```

Fescron, 2025-02-11:
Alright, running the "one liner" took quite some time but now everything works as expected, thank you!

The only thing I see that still might not be correct: there is a "square" in front of the `ttyUSBx@115200` status field on the bottom right. I suppose that symbol is still missing somehow.

Thanks again for all the help and the (so far) nice product! Until now I've done it on a "test" Ubuntu laptop but I suppose I can now run it on my main device :) Although I do look forward to perhaps a "portable" Ubuntu (and Windows) application :)

nvtkaszpir, 2025-04-22:
looks like build script was moved to https://github.com/TechxArtisanStudio/Openterface_QT/blob/main/build-script/build-from-source.sh