TechxArtisanStudio/Openterface_QT #127: Maximize behavior on Windows is incorrect

ghjm, 2025-02-01:
When I click the maximize button on the Windows client, instead of maximizing, it resizes to fill the top part of the display.  Once this has happened, the only way to get it back to the original non-maximized layout is to change the target screen resolution.  So there are two problems related to this:

* Maximizing should maximize the window as per a normal Windows app
* Once maximized, un-maximizing should return the window to its previous state

rvcjew, 2025-02-02:
Also when this happens it cuts off the bottom of the display window by like 30 pixels.

akadaedalus, 2025-02-19:
I actually see this behavior in a tiling manager (awesome wm) in Linux as well.  The bottom status bar is off screen.

closed 2025-04-02 by Chow411