TechxArtisanStudio/Openterface_QT #129: Debian 12: libstdc++.so.6: version `GLIBCXX_3.4.32' not found

tunloop, 2025-02-01:
Looks like the binary in the `.deb` package requires a version that is not supported by the most current Debian stable release.

```
# ldd /usr/bin/openterfaceQT
/usr/bin/openterfaceQT: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /usr/bin/openterfaceQT)
```

Debian stable supported versions:
```
# strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_3.4.29
GLIBCXX_3.4.30
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

kevinzjpeng, 2025-02-01:
Thanks for reporting this issue. I think the root cause of the package that using the "GLIBCXX_3.4.32" is because I used the github action latest ubuntu to build the openterfaceQT package, I will try to lower the ubuntu version to build it. But it take some time, as you know resolving the dependency of packages is time consuming. 

On another hand, if you can build the openterfaceQT from source, it will be a quick work around. 

I will update here once I can lower the GLIBCXX version.

kevinzjpeng, 2025-02-01:
After investigating the issue, I have identified the root cause and implemented a solution. Here's a summary of the findings and the steps taken:

#### Root Cause:
The error occurred because the binary `/usr/bin/openterfaceQT` requires `libstdc++.so.6` version `GLIBCXX_3.4.32`, which corresponds to GCC 13.2.0 (as per the [GCC ABI documentation](https://gcc.gnu.org/onlinedocs/libstdc++/manual/abi.html)). However, the current Debian stable release only supports up to `GLIBCXX_3.4.30`, as it includes an older version of `libstdc++`.

This discrepancy arose because the project's GitHub Actions runner was using **Ubuntu 24.04**, which ships with GCC 13.2.0 (`4:13.2.0-7ubuntu1`). As a result, the binary was built against this newer version of `libstdc++`, making it incompatible with systems using older library versions.

#### Solution:
To resolve this compatibility issue, I downgraded the GitHub Actions runner to **Ubuntu 22.04**, which uses GCC 11.2.0 (`4:11.2.0-1ubuntu1`). This version of GCC corresponds to `libstdc++.so` versions in the range of **6.0.29–6.0.30**, which aligns with the library versions available in the Debian stable release.

By making this change, the binary will now be built against a version of `libstdc++` that is compatible with Debian stable, ensuring smooth execution without runtime errors.

---

Can you please help to test the package under https://github.com/TechxArtisanStudio/Openterface_QT/actions/runs/13088097594
If it works well, I will keep using **Ubuntu 22.04** as build runner.

tunloop, 2025-02-01:
Yes, that is working now. Thank you very much!

closed 2025-02-02 by kevinzjpeng