TechxArtisanStudio/Openterface_QT #131: Feature request: Portable Windows host app

bret-j, 2025-02-01:
Since the Openterface is touted as a portable toolkit, I would very much like to see a portable Windows host app that does **not** require installation.  I understand it would still need administrator privileges since it includes driver(s), etc. but open source software VeraCrypt also does that and it has an official portable version.  Having the ability to run the app from a USB flash drive would really make this product handy and portable.

Thanks for your consideration.

kevinzjpeng, 2025-02-14:
I’m excited to share an update regarding the portable version of the app. After careful consideration and effort, I’ve successfully developed a solution to build the app into a single executable for Windows. This portable version will be included in the next release. Stay tuned for more details!

Chow411, 2025-06-11:
Lastest version have portable windows host app

closed 2025-06-11 by Chow411