TechxArtisanStudio/Openterface_QT #135: Video and Keyboard problems

DOMZE, 2025-02-04:
Hello,

Just received my module this afternoon. After testing it out on my Dell R430, here are the problems I encountered:

1. Keyboard does not work. I have the VGA to HDMI adapter + the orange cable USB-C to USB-C+A adapter and I cannot get the keyboard to work, whether it's me using my physical keyboard or using the integrated keyboard in the application. I even tried switching the USB port around, no dice.
2. The video is not shown for the following output. The screen is both the app the from the iDRAC app.

[image]
[image]

The video is not so problematic, as I can actually see what is happening once that screen boots. But the keyboard is problematic. I can't do nothing.

Please advise.

CableCatDK, 2025-02-04:
For me video has worked surprisingly well. So I have no suggestions there.

For the keyboard I have struggled a lot, there are two place where the connection can break:

1. The virtual serial port between host and openterface. This it not the case for you, as you see COM4@115200
2. The USB/HID connection between openterface and target. This is your issue, because you see the mouse and keyboard icons are red.

I mostly have the last issue, like you. To fix it one or more are required:
- Reset the virtual serial port from the app.
- Disconnect the USB connection between openterface and target.
- Close the openterface app, and disconnect beteen openterface and host. Restart and replug.

The manufacture recomments that you do not use the orange USB-C cable. But instead the high quality USB-C to USB-C cable.

kevinzjpeng, 2025-02-04:
> Hello,
> 
> Just received my module this afternoon. After testing it out on my Dell R430, here are the problems I encountered:
> 
> 1. Keyboard does not work. I have the VGA to HDMI adapter + the orange cable USB-C to USB-C+A adapter and I cannot get the keyboard to work, whether it's me using my physical keyboard or using the integrated keyboard in the application. I even tried switching the USB port around, no dice.
> 2. The video is not shown for the following output. The screen is both the app the from the iDRAC app.
> 
> <img alt="Image" width="962" src="https://private-user-images.githubusercontent.com/8238156/409356568-c69eca6c-ad6a-45a0-8bf7-b4b79c3a8afa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg2NjY3NjUsIm5iZiI6MTczODY2NjQ2NSwicGF0aCI6Ii84MjM4MTU2LzQwOTM1NjU2OC1jNjllY2E2Yy1hZDZhLTQ1YTAtOGJmNy1iNGI3OWMzYThhZmEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAyMDRUMTA1NDI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MmYzMmI2OWFkNWY0NjRiYzRlOTA3NzgzNmRkY2ZkNWRjZTFhYzNmY2QxMTkyNTRiYWNkYjQ5OWViN2M2YmMyZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.J2-LYluau4TyzpSivl28SzSXTAu9Lf4-svlo6pb9YEw"> <img alt="Image" width="769" src="https://private-user-images.githubusercontent.com/8238156/409356569-c2779041-40e7-4a9c-a6ab-d4567efb341d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg2NjY3NjUsIm5iZiI6MTczODY2NjQ2NSwicGF0aCI6Ii84MjM4MTU2LzQwOTM1NjU2OS1jMjc3OTA0MS00MGU3LTRhOWMtYTZhYi1kNDU2N2VmYjM0MWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAyMDRUMTA1NDI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmRhZmNmY2VmODRkMTI3ZWQ1MmYxMDU4YWZlZWYyYTA2Njc5NjQ0MjViMTU2NzRmMGZmOGY2MDA5MmNlNzIyZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Vgpc42DngeLN07WSVy1v7lym-SJlNKVYmKj6lBXnQ3k">
> The video is not so problematic, as I can actually see what is happening once that screen boots. But the keyboard is problematic. I can't do nothing.
> 
> Please advise.

Hi @DOMZE ,  

Thank you for the detailed report. Let’s work through these issues step by step:  

1. **Video Output Issues**  
   - First, could you confirm whether you’re using the **official VGA-to-HDMI adapter from Openterface**, or a third-party VGA-to-HDMI adapter?  
   - If you’re using the **Openterface VGA-to-HDMI adapter**, please ensure the **USB cable is plugged into a power source**. This can be either the target USB port or the Openterface USB Type-A port. The adapter requires external power to function properly.  
   - If the wiring is correct, I suggest isolating the issue by:  
     - Testing the Openterface module and software with another computer that has HDMI output.  
     - Testing the VGA-to-HDMI adapter with a different HDMI display.  
   - Regarding the screen issue you mentioned (visible during boot but not afterward), this could be due to a resolution change after boot. The subsequent resolution might not be supported by the VGA-to-HDMI adapter. If possible, try setting the system to a standard resolution (e.g., 1024x768 or 1280x720) for testing.  

2. **Keyboard Not Working**  
   - Try switching the **baud rate to 9600** by navigating to:  
     - **Menu > Control > Baudrate > 9600**.  
   - If the keyboard and mouse start working, it indicates the HID chip may have been reset for some reason. To resolve this:  
     - Unplug all cables.  
     - Reconnect only the host with the Openterface app open. The system should automatically correct the baud rate.  

Let me know the results of these tests, and we’ll proceed from there. If the issues persist, we can explore further troubleshooting steps.

DOMZE, 2025-02-05:
Hi @CableCatDK and @kevinzjpeng 

Here are some more information:

**Video Output Issues**

1. I am indeed using the official VGA-to-HDMI adapter from Openterface.
2. The video output works. It's just at a certain stage (stage shown from iDRAC below) that I don't see nothing. Once that one is done, everything works fine. I changed the resolution to 1024x768@60hz and still no dice. I even went to 800x600, still not working.

it only occurs when it's loading this stuff:

[image]
[image]

<br/>**Keyboard Not Working**

All the following steps were used using the official 30cm USB-C to USB-C cable. For my scenario I am using USB-C (on KVM) to USB-A (on target)

Change the baud to 9600 did not do anything.
This worked:

1. Reset the virtual serial port from the app.
2. Disconnect the USB connection between openterface and target.
3. Close the openterface app and restart the openterface app.

kevinzjpeng, 2025-02-11:
Hi @DOMZE, Based on the symptoms described and the iDRAC interface's reported frame rates (5.0fps and 15.2fps), it appears that the root cause may be related to the VGA-to-HDMI adapter's compatibility with certain refresh rates.  

The chipset inside your adapter, **MS9288C**, has specific limitations regarding supported resolutions and refresh rates. Unfortunately, the frame rates displayed in your iDRAC interface (5.0fps and 15.2fps) are not among the officially supported configurations for this chipset. This mismatch can lead to video output issues during certain stages of the boot process or firmware initialization.  

To assist you further, I’ve compiled a list of the **MS9288C-supported resolutions and refresh rates** for your reference:  
- 640x480 : 60 Hz, 75 Hz
- 800x600 : 60 Hz, 75 Hz
- 1024x768 : 60 Hz, 75 Hz
- 1152x864 : 60 Hz, 75 Hz
- 1280x600 : 60 Hz
- 1280x720 : 60 Hz
- 1280x768 : 60 Hz
- 1280x800 : 60 Hz, 75 Hz
- 1280x960 : 60 Hz
- 1280x1024 : 60 Hz, 75 Hz
- 1360x768 : 60 Hz
- 1366x768 : 60 Hz
- 1400x1050 : 60 Hz
- 1440x900 : 60 Hz
- 1600x900 : 60 Hz
- 1600x1200 : 60 Hz
- 1680x1050 : 60 Hz
- 1920x1080 : 60 Hz
- 1920x1200 : 60 Hz
- 480i : Interlaced
- 480p : Progressive
- 576i : Interlaced
- 576p : Progressive
- 720p : 50 Hz, 60 Hz
- 1080i : 50 Hz, 60 Hz
- 1080p : 50 Hz, 60 Hz

### Next Steps:  
1. **Test with Supported Settings:**  
   Please ensure that the resolution and refresh rate settings on your target device align with the supported configurations of the MS9288C chipset. For example, try setting the resolution to **1024x768 @ 60Hz** (or another supported mode) and check if the issue persists.  

2. **Test with a Different Adapter:**  
   If possible, test the setup with a different VGA-to-HDMI adapter or a direct HDMI connection to confirm whether the issue is specific to the current adapter.  

We understand how critical it is to have a stable and reliable setup, and we’re committed to helping you resolve this issue. Please don’t hesitate to reach out if you need additional assistance or clarification.

DOMZE, 2025-03-09:
Hi @kevinzjpeng,

As mentioned in my earlier post, I did try to change the resolutions to a few ones, including **1024x768@60hz**. This did not work. I understand that the chipset has limitations and that's great. It would be good for you and the team to document those limitations in your documentation so that other folks that are looking to buy directly from you understand the limitations.

In regards to testing with another adapter, I do not have one handy.