TechxArtisanStudio/Openterface_QT #136: Please add support for 1920x1200 resolution

zorxd, 2025-02-04:
Hi

Using latest windows host app (0.0.6 reported as 0.0.39.009)

Under file->preference->video->capture resolutions

There are a limited number of choices including 1920x1080 but NOT 1920x1200 or any other 16:10 resolution.

This causes issues on my laptop with black bars on the side when using "clone" as the video output and an offset on the mouse.
The simple solution would be too add more resolution choices.

[image]

[image]

CableCatDK, 2025-02-04:
This is a limitation of the hardware. You can use another capturecard, and get other resolutions.

zorxd, 2025-02-04:
OK I understand the hardware must have a maximum resolution (but maybe the CPU is the limiting factor here and 1920x1200 could work with a lower refresh rate or color depth).

However, it's not an excuse not to support other 16:10 resolutions such as say, 1600x900.

kevinzjpeng, 2025-02-11:
@zorxd Thank you for raising this issue. We understand that the lack of 16:10 resolutions, such as 1920x1200, in the "clone" mode causes black bars on the sides and mouse offset problems on your laptop. Unfortunately, this is a hardware limitation of the capture chip, which does not support these resolutions natively.

While we cannot change the hardware constraints, we are exploring software-based improvements to enhance the user experience for 16:10 displays. One potential approach is to automatically hide or adapt the black areas when using "clone" mode to provide a cleaner and more seamless experience.

We appreciate your feedback and will continue to work on making the software more adaptable to different display setups.

zorxd, 2025-06-12:
mouse is now OK. There are still useless black bars on the side however.

[image]