TechxArtisanStudio/Openterface_QT #137: Keyboard/mouse not working on Ubuntu 24.04

jsmarion, 2025-02-05:
Hi - I am having a similar issue to others I see here. I am connected to a RPi5. I get good video, but the keyboard/mouse don't work on the RPi. My software says it is 0.0.39.009 and if I check it says an update is available, but takes me to the same software I have installed which is shown on the site as 0.0.6. Also, if I try to set the baud rate, I get a segfault and the app crashes. Any ideas? Thanks.

kevinzjpeng, 2025-02-11:
Hello,

Thank you for reaching out. To better assist you for the Keyboard and mouse issue, could you please provide some additional information?

1. **Environment Information**:  
   Open the Openterface QT app, navigate to `Help -> Environment`, and copy the environment information here.

2. **Driver Check**:  
   - If you're using the Windows version, please check the Device Manager to ensure that the serial port driver is properly installed.
   - If you're on Linux, open a terminal and run the following command:
     ```
     lsusb -t
     ```
     Please share the output with me.

This information will help us diagnose the issue more effectively. Additionally, regarding the baud rate setting causing a segmentation fault, we'll look into this as well once we have the above details.

Thank you for your cooperation!

jsmarion, 2025-02-11:
Thanks for your response. I am on Ubuntu 24.04. The environment is:

> App: 0.0.39.009
> OS: Ubuntu 24.04.1 LTS
> QT: 6.4.2
> 
> Permissions:
> Microphone Available
> VideoA vailable
> 
> Environment Variables:
> QT_QPA_PLATFORM: xcb
> XDG_SESSION_TYPE: x11
> WAYLAND_DISPLAY: (not set)
> DISPLAY: :1

The results of 'lsusb -t':

> /:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 480M
>     |__ Port 001: Dev 002, If 0, Class=Human Interface Device, Driver=usbhid, 12M
>     |__ Port 006: Dev 003, If 0, Class=Hub, Driver=hub/4p, 480M
> /:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
> /:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 480M
>     |__ Port 002: Dev 002, If 0, Class=Vendor Specific Class, Driver=[none], 12M
>     |__ Port 005: Dev 003, If 0, Class=Wireless, Driver=btusb, 12M
>     |__ Port 005: Dev 003, If 1, Class=Wireless, Driver=btusb, 12M
>     |__ Port 006: Dev 004, If 0, Class=Hub, Driver=hub/4p, 480M
>         |__ Port 003: Dev 005, If 0, Class=Hub, Driver=hub/4p, 480M
>             |__ Port 001: Dev 006, If 0, Class=Hub, Driver=hub/4p, 480M
>                 |__ Port 001: Dev 040, If 0, Class=Hub, Driver=hub/4p, 480M
>                     |__ Port 001: Dev 041, If 0, Class=Video, Driver=uvcvideo, 480M
>                     |__ Port 001: Dev 041, If 1, Class=Video, Driver=uvcvideo, 480M
>                     |__ Port 001: Dev 041, If 2, Class=Audio, Driver=snd-usb-audio, 480M
>                     |__ Port 001: Dev 041, If 3, Class=Audio, Driver=snd-usb-audio, 480M
>                     |__ Port 001: Dev 041, If 4, Class=Human Interface Device, Driver=usbhid, 480M
>                     |__ Port 002: Dev 042, If 0, Class=Vendor Specific Class, Driver=[none], 12M
>                 |__ Port 003: Dev 008, If 0, Class=Human Interface Device, Driver=usbhid, 12M
>                 |__ Port 003: Dev 008, If 1, Class=Human Interface Device, Driver=usbhid, 12M
>                 |__ Port 003: Dev 008, If 2, Class=Chip/SmartCard, Driver=[none], 12M
>                 |__ Port 004: Dev 039, If 0, Class=Communications, Driver=cdc_acm, 12M
>                 |__ Port 004: Dev 039, If 1, Class=CDC Data, Driver=cdc_acm, 12M
>                 |__ Port 004: Dev 039, If 2, Class=Mass Storage, Driver=usb-storage, 12M
>                 |__ Port 004: Dev 039, If 3, Class=Human Interface Device, Driver=usbhid, 12M
>                 |__ Port 004: Dev 039, If 4, Class=Audio, Driver=snd-usb-audio, 12M
>                 |__ Port 004: Dev 039, If 5, Class=Audio, Driver=snd-usb-audio, 12M
>             |__ Port 002: Dev 007, If 0, Class=Human Interface Device, Driver=usbhid, 12M
>             |__ Port 002: Dev 007, If 1, Class=Human Interface Device, Driver=usbhid, 12M
>             |__ Port 002: Dev 007, If 2, Class=Human Interface Device, Driver=usbhid, 12M
>             |__ Port 004: Dev 009, If 0, Class=Audio, Driver=snd-usb-audio, 12M
>             |__ Port 004: Dev 009, If 1, Class=Audio, Driver=snd-usb-audio, 12M
>             |__ Port 004: Dev 009, If 2, Class=Audio, Driver=snd-usb-audio, 12M
>             |__ Port 004: Dev 009, If 3, Class=Human Interface Device, Driver=usbhid, 12M
> /:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
> /:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 480M
>     |__ Port 001: Dev 002, If 0, Class=Human Interface Device, Driver=usbhid, 12M
> /:  Bus 006.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M

I appreciate your help.

jsmarion, 2025-03-03:
Hi - can you please help with this @kevinzjpeng ? Thanks.

jsmarion, 2025-03-19:
@kevinzjpeng - can you please help?

silentyak, 2025-04-02:
I'm on Windows and I have the same issue, my keyboard and mouse is not working. I had to install the drivers from https://github.com/TechxArtisanStudio/Openterface_QT/issues/138 to get it to connect, now I get video but no mouse/keyboard support. 

I'm not sure what I need to look for in Device Manager for `ensure that the serial port driver is properly installed.`

Environment:
```
App: 0.3.0.087
OS: Windows 11 Version 24H2
QT: 6.5.3

Permissions:
Microphone Available
Video Available

Environment Variables:
QT_QPA_PLATFORM: (not set)
XDG_SESSION_TYPE: (not set)
WAYLAND_DISPLAY: (not set)
DISPLAY: (not set)
```

jsmarion, 2025-04-07:
I guess we will just hot get any support for this. It's disappointing.

Chow411, 2025-06-11:
We have added driver checking functionality to our software. Please download the latest version and give it a try.

Chow411, 2025-06-12:
Need add this to /etc/udev/rules.d/51-openterface.rules udev get the HID permission
```
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="534d", ATTRS{idProduct}=="2109", MODE="0666"
```

then run follow command or reboot
``` sh
sudo udevadm control --reload-rules
sudo udevadm trigger
```

closed 2025-06-12 by jsmarion