TechxArtisanStudio/Openterface_QT #138: Driver installation issue on some Windows 11 machines

kashalls, 2025-02-05:
### The Issue

Some windows machines during the Openterface installer will fail to install the required drivers due to an unknown reason.
Pulling the drivers from [this repo](https://github.com/TechxArtisanStudio/Openterface_QT/tree/main/driver/windows) results in Windows throwing the "system cannot find the field specified".

[image]

### Known Affected Machines
[image]

### Potential Workaround / Fix
The current working fix is to pull the drivers from https://wch-ic.com/downloads/CH341SER_ZIP.html and pointing Window's device manager to use those, which is working in 22H2 and up with Secure Boot and Requiring Driver Signing.

kevinzjpeng, 2025-02-06:
After investigating the driver issue, we discovered that we are currently using an older version of the WCH driver (v3.8). However, a newer version (v3.9) was released on September 16, 2024, which is more recent than the version used in our Openterface_QT development.

I came across some useful insights from this Stack Overflow thread: [Serial to USB Cable CH340 New Driver Not Functioning with C# Application](https://stackoverflow.com/questions/76158517/serial-to-usb-cable-ch340-new-driver-not-functioning-with-c-sharp-applicatio) . Based on the information provided, the root cause of the issue likely stems from compatibility problems with WCH driver version 3.8.2023.2, particularly on Windows 22H2 and later versions. This is the same driver version currently used in Openterface_QT.

To address this, I have upgraded the WCH driver to version 3.9 in the installer, which should ideally resolve the issue. However, since I do not have access to a system running the problematic version of Windows, I would greatly appreciate it if any user could help test this updated build on such an environment. Please let me know if the issue persists after the upgrade.

The latest build link: https://github.com/TechxArtisanStudio/Openterface_QT/actions/runs/13177576720

Roxedus, 2025-02-06:
Running the linked installer does not install the driver, but a manual install with the files from the 3.9 folder does work.

kevinzjpeng, 2025-02-08:
> Running the linked installer does not install the driver, but a manual install with the files from the 3.9 folder does work.

Thank you for testing! Previously, we used rundll32 setupapi,InstallHinfSection to install the CH340B driver, but we encountered compatibility issues with certain Windows releases. To address this, we have now switched to using pnputil.exe for driver installation, which we hope will provide better compatibility across different Windows versions.

You can find the latest build here: https://github.com/TechxArtisanStudio/Openterface_QT/actions/runs/13213997671

Roxedus, 2025-02-08:
That seems to do the trick :)

kevinzjpeng, 2025-02-11:
New release should be able to address this issue, https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/v0.1.0

I will close this issue.

closed 2025-02-11 by kevinzjpeng