TechxArtisanStudio/Openterface_QT #139: Keyboard / Layout Issue German Keyboard / Systems

skuehnemundt, 2025-02-06:
Hi,
sry i opend my issue on the wrong repo first: https://github.com/TechxArtisanStudio/Openterface/issues/12 

I received my minikvm last week, when testing it for the first time I noticed a problem

I am sitting here on 2 Thinkpads with German keyboards, when I have the KVM in between to remotely manage a notebook, the expected keys do not come back. Is there any way to set the layout you are working with or any configuration for it? I haven't found any information about this on your website yet.

On the host, I type a # on the German keyboard and a 3 comes up on the target. NUM-Lock on/off did not help.

App: 0.0.39.009OS: Ubuntu 24.04.1 LTSQT: 6.4.2

Permissions:
Microphone Available
Video Available

Environment Variables:
QT_QPA_PLATFORM: xcb
XDG_SESSION_TYPE: x11
WAYLAND_DISPLAY: (not set)
DISPLAY: :0

the app shows correctly that I type an “ö” on the target, but nothing arrives. When pressing “,” as expected, “,” appears with “shift + ,” a “;” should appear, but “Ö” is passed on
I have also attached a log with the keystrokes “ ö Ö , ; ”.

[openterface_log_keyboard.txt](https://github.com/user-attachments/files/18687831/openterface_log_keyboard.txt)

CableCatDK, 2025-02-06:
The current workaround is to set the host keyboard to US for the MiniKVM app. Symbols that needs AltGr, still does not work.

Future version will allow you the set the target keyboard.

I would preffered that the keyboard scancode where passed directly from host to target.

Chow411, 2025-06-11:
We have fixed many keyboard issues related to German language input in the latest version. Please give it a try.