TechxArtisanStudio/Openterface_QT #141: Keyboard/Mouse may not work if other USB-Serial Adapters are present

slohse, 2025-02-06:
The checks in [SerialPortManager](https://github.com/TechxArtisanStudio/Openterface_QT/blob/79613d4c2b716b0a695f5b9f6154ff1c3f2bcdcd/serial/SerialPortManager.cpp#L77) are not sufficient to conclusively determine which serial port belongs to Openterface. I happened to have another device with a CH340 serial converter connected to my computer when I tested Openterface and this kept Keyboard and Mouse from working. The relevant output looked like this (`ttyUSB1` is the serial interface provided by the Openterface hardware): 

```
{Debug}: Check serial port.
{Debug}: Serial port connected:  "ttyUSB1" baudrate: 115200
{Debug}: Open port "ttyUSB1, baudrate: " 115200
{Debug}: Check sum "57 ab 00 08 00 0a"
{Debug}: "Data written to serial port: @ttyUSB1" : "57 ab 00 08 00 0a"
{Debug}: Data read from serial port:  "57 ab 00 88 32 82 80 00 00 01 c2 00 08 00 00 03 86 1a 29 e1 00 00 00 01 00 0d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 44"
{Debug}: prefix: "57ab" | addr1: 0 | cmd: "88" | len: 50 | mode: "82" | cfg: "80" | addr2: "0" | baudrate: 115200 | reserved1: "800" | serial_interval: 3 | vid: "1a86" | pid: "e129" | keyboard_upload_interval: 0 | keyboard_release_timeout: 1 | keyboard_auto_enter: 0 | enterkey1: "0" | enterkey2: "0" | filter_start: "0" | filter_end: "0" | custom_usb_desc: 0 | speed_mode: 0 | reserved2: "0" | reserved3: "0" | reserved4: "0" | sum: "0"
{Debug}: Check serial port completed.
{Debug}: Serial port connection success:  "ttyUSB1"
{Debug}: Observe "ttyUSB1" data ready and bytes written.
{Debug}: Enable the switchable USB now...
{Debug}: Check sum "57 ab 00 01 00 03"
{Debug}: "Data written to serial port: @ttyUSB1" : "57 ab 00 01 00 03"
{Debug}: Receive from serial port @ 115200 : "57 ab 00 81 08 38 01 00 00 00 00 00 00 c4"
{Debug}: Data recived:  0
{Debug}: Serial port connected:  "ttyUSB0" baudrate: 115200
{Debug}: Serial port is already opened.
{Debug}: Check sum "57 ab 00 08 00 0a"
{Debug}: "Data written to serial port: @ttyUSB1" : "57 ab 00 08 00 0a"
{Debug}: Receive from serial port @ 115200 : "57 ab 00 88 32 82 80 00 00 01 c2 00 08 00 00 03 86 1a 29 e1 00 00 00 01 00 0d 00 00 00 00 00 00"
{Debug}: Current serial port baudrate rate: 115200 , Mode: "0x82"
{Debug}: Serial is ready for communication.
{Debug}: Baud rate is already set to 115200
{Debug}: Receive from serial port @ 115200 : "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 44"
{Debug}: Unknown command:  "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 44"
{Debug}: No data with 115200 baudrate, try to connect:  "ttyUSB0" with baudrate: 9600
{Debug}: Close serial port
{Debug}: Unregister obseration of the data ready
{Debug}: Open port "ttyUSB0, baudrate: " 9600
{Debug}: Check sum "57 ab 00 08 00 0a"
{Debug}: "Data written to serial port: @ttyUSB0" : "57 ab 00 08 00 0a"
{Debug}: Data read from serial port with 9600:  ""
{Debug}: Check serial port completed.
{Debug}: Serial port connection success:  "ttyUSB0"
{Debug}: Observe "ttyUSB0" data ready and bytes written.
{Debug}: Enable the switchable USB now...
{Debug}: Check sum "57 ab 00 01 00 03"
{Debug}: "Data written to serial port: @ttyUSB0" : "57 ab 00 01 00 03"
```

My interpretation of this is: Despite successfully communicating with Openterface over `ttyUSB1`, the software then uses `ttyUSB0` for some reason.

Now, I'm neither experienced with QT, nor have I ever worked with libusb or know how the innards of USB work. But shouldn't it be somehow possible to find the serial interface that is on the same port as the video and audio devices that Openterface provide?

kevinzjpeng, 2025-02-10:
Thank you for bringing this issue to our attention. We understand that when multiple CH340 devices are connected, the Openterface software may incorrectly communicate with the wrong serial port, leading to issues like keyboard and mouse functionality being disrupted.

#### Current Workaround:
To ensure proper functionality, we recommend connecting the Openterface device **first** before plugging in any other CH340 devices. This allows the software to correctly identify and communicate with the Openterface hardware.

#### Why This Happens:
The operating system assigns generic names (e.g., `ttyUSB0`, `ttyUSB1`) to all CH340 devices, making it difficult for the software to distinguish between them. Currently, the software does not have a robust way to uniquely identify the Openterface device when multiple CH340 devices are present.

#### What We’re Doing to Fix It:
We are actively working on improving the software to:
- Store the unique identifier of the Openterface device during initialization.
- Ignore other CH340 devices that are not part of the Openterface KVM system.

This enhancement will ensure that the software reliably communicates with the correct device, even when multiple CH340 devices are connected. We will update you here once the improvements are implemented.

ZeroKnight, 2025-03-13:
I believe that you'll need to walk through the device tree in order to sufficiently determine the correct serial device; you should be able to do it with libusb.

The most reliable way I've found of identifying the Openterface so far is by looking for the HID product name as you're already doing [here](https://github.com/TechxArtisanStudio/Openterface_QT/blob/2e641c1fe55c98f07e72d8350cbae03f575f9dfb/video/videohid.cpp#L427), or by its vendor/product IDs as I've done in #203. I believe that the *specific* device identified here is the HDMI capture card.

By using `udevadm info`, I can get a device tree based on this device:

`udevadm info --attribute-walk --name /dev/hidraw9`
<details>
<summary>Output</summary>

    looking at device '/devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-1/3-1.1/3-1.1:1.4/0003:534D:2109.0010/hidraw/hidraw9':
      KERNEL=="hidraw9"
      SUBSYSTEM=="hidraw"
      DRIVER==""
      ATTR{power/async}=="disabled"
      ATTR{power/control}=="auto"
      ATTR{power/runtime_active_kids}=="0"
      ATTR{power/runtime_active_time}=="0"
      ATTR{power/runtime_enabled}=="disabled"
      ATTR{power/runtime_status}=="unsupported"
      ATTR{power/runtime_suspended_time}=="0"
      ATTR{power/runtime_usage}=="0"

    looking at parent device '/devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-1/3-1.1/3-1.1:1.4/0003:534D:2109.0010':
      KERNELS=="0003:534D:2109.0010"
      SUBSYSTEMS=="hid"
      DRIVERS=="hid-generic"
      ATTRS{country}=="21"
      ATTRS{power/async}=="enabled"
      ATTRS{power/control}=="auto"
      ATTRS{power/runtime_active_kids}=="0"
      ATTRS{power/runtime_active_time}=="0"
      ATTRS{power/runtime_enabled}=="disabled"
      ATTRS{power/runtime_status}=="unsupported"
      ATTRS{power/runtime_suspended_time}=="0"
      ATTRS{power/runtime_usage}=="0"

    looking at parent device '/devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-1/3-1.1/3-1.1:1.4':
      KERNELS=="3-1.1:1.4"
      SUBSYSTEMS=="usb"
      DRIVERS=="usbhid"
      ATTRS{authorized}=="1"
      ATTRS{bAlternateSetting}==" 0"
      ATTRS{bInterfaceClass}=="03"
      ATTRS{bInterfaceNumber}=="04"
      ATTRS{bInterfaceProtocol}=="00"
      ATTRS{bInterfaceSubClass}=="00"
      ATTRS{bNumEndpoints}=="01"
      ATTRS{power/async}=="enabled"
      ATTRS{power/runtime_active_kids}=="0"
      ATTRS{power/runtime_enabled}=="enabled"
      ATTRS{power/runtime_status}=="suspended"
      ATTRS{power/runtime_usage}=="0"
      ATTRS{supports_autosuspend}=="1"

    looking at parent device '/devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-1/3-1.1':
      KERNELS=="3-1.1"
      SUBSYSTEMS=="usb"
      DRIVERS=="usb"
      ATTRS{authorized}=="1"
      ATTRS{avoid_reset_quirk}=="0"
      ATTRS{bConfigurationValue}=="1"
      ATTRS{bDeviceClass}=="ef"
      ATTRS{bDeviceProtocol}=="01"
      ATTRS{bDeviceSubClass}=="02"
      ATTRS{bMaxPacketSize0}=="64"
      ATTRS{bMaxPower}=="500mA"
      ATTRS{bNumConfigurations}=="1"
      ATTRS{bNumInterfaces}==" 5"
      ATTRS{bcdDevice}=="2100"
[... 246 lines cut ...]
      ATTRS{power/wakeup_expire_count}==""
      ATTRS{power/wakeup_last_time_ms}==""
      ATTRS{power/wakeup_max_time_ms}==""
      ATTRS{power/wakeup_total_time_ms}==""
      ATTRS{product}=="USB2.0 HUB"
      ATTRS{quirks}=="0x4000"
      ATTRS{removable}=="removable"
      ATTRS{remove}=="(not readable)"
      ATTRS{rx_lanes}=="1"
      ATTRS{speed}=="480"
      ATTRS{tx_lanes}=="1"
      ATTRS{urbnum}=="38"
      ATTRS{version}==" 2.00"

    [snip]

</details>

Comparing the output of these two device trees, we can see that the last device in that output is the USB Hub that the both HID device and the CH341 serial device are connected to (**3-1** in my case). So, if you're able to get to this parent device, you can then enumerate its child devices to find the correct serial device.

Unfortunately, we can't make a udev rule based on this information since the serial device is a sibling to the HID device and and udev rules can only match *up* the tree.

I don't know if there's a Qt module for USB devices, but since you've already pulled in **libusb**, you could use its interface; `libusb_get_parent()` and `libusb_get_device_list()` seem promising.

You could also use [libudev](https://github.com/systemd/systemd/tree/main/src/libudev) or its successor, [sd-device](https://github.com/systemd/systemd/blob/main/src/systemd/sd-device.h), but either of those will limit you to systemd-based systems.