TechxArtisanStudio/Openterface_QT #148: Key presses are skipped, if mouse is moved

CableCatDK, 2025-02-08:
Mouse movements flood the serial interface. This causes key presses to be skipped. It also seems like mouse movements are weird because of it.
I suggest that key presses and mouse clicks are prioritized before mouse movements. Even better if mouse movements are limited to the speed of the erial interface.

zorxd, 2025-02-11:
there seem to be a rate limit on the amount of keys that can be sent. If I press SHIFT + three letters (to write something in capitals such as USA), only 2 letters will get through (typically first and last letter, so in that case I get UA instead of USA)

Chow411, 2025-06-11:
We have optimized the serial communication, which will greatly reduce this issue. We are closing this issue now, but please remind us to reopen if there are still problems.

closed 2025-06-11 by Chow411