TechxArtisanStudio/Openterface_QT #169: More packaging options for Linux (rpm, AppImage, Flatpak)

HubbeKing, 2025-02-14:
It would be very nice to have other options beyond a .deb for the Linux version.
Packaging an .rpm as well for Fedora support would likely not be much work.

Another possibility is something like AppImage, which would be distribution-agnostic and work everywhere.
I'm guessing a Flatpak or Snap would be difficult, however, due to the permissions needed for serial and HIDraw access.

xnfra, 2025-02-16:
I came here to say the same thing. I'm trying to build my own local AppImage from source but I would prefer it being a Flatpak on flathub. I do know flatpaks need to run with the option of `--device=all` to allow serial access. An AppImage and offical flatpak would be great!

nvtkaszpir, 2025-04-22:
under Ubuntu 22.04 I had issues with Appimage, but flatpark works flawlessly, thanks!

Chow411, 2025-06-11:
Lastest version added rpm, AppImage, Flatpak.

closed 2025-06-11 by Chow411