TechxArtisanStudio/Openterface_QT #170: High CPU usage when nothing is happening

dodancs, 2025-02-14:
I noticed that one of my CPU cores is constantly pinned to 90-100%, then when I opened htop, I saw that OpenterfaceQT was constantly using a lot of CPU:

[image]

Is this expected behavior? Maybe this is ffmpeg or something in the background transcoding the video? Or is it a bug?

kevinzjpeng, 2025-02-18:
Thank you for reporting this issue. In our testing on Ubuntu 24.04, CPU usage for OpenterfaceQT remains normal, so I’m unable to reproduce the problem. Could you please share your system specifications (OS version, CPU, GPU, RAM, etc.) and any additional details about your setup? This will help us investigate further and determine if this is expected behavior or a potential bug.

mvirkkunen, 2025-02-18:
For what it's worth, I see about 70% CPU usage when the app is sitting idle on a mid range laptop from a few years ago. Even when "nothing is happening", as long as the video feed is visible, it does still have to decode the webcam video feed in real time which takes some CPU time (it's an MJPEG stream so it's essentially an endless series of JPEGs).

kevinzjpeng, 2025-02-18:
The simplest way to determine whether the high CPU usage is caused by an Openterface_QT bug or by the video feed itself is to open a popular webcam player, such as VLC Player. If you notice that the CPU usage is also high in VLC, it likely means that your laptop is not processing the video feed properly. This could be due to improperly installed graphics card drivers.

For reference, I am using an older mini PC with an i7-9700 @ 3.00GHz processor and an NVIDIA GTX 1050 GPU, and it runs at around 10-15% CPU usage on a single core under similar conditions.

dodancs, 2025-02-26:
> Could you please share your system specifications (OS version, CPU, GPU, RAM, etc.)

Laptop: Lenovo Thinkpad T14s gen4 (AMD)
CPU: AMD Ryzen 7 PRO 7840U
GPU: Radeon 780M Graphics (integrated)
RAM: 32GB DDR5
OS: Ubuntu 24.04.2 LTS

dodancs, 2025-02-26:
> If you notice that the CPU usage is also high in VLC, it likely means that your laptop is not processing the video feed properly.

When OpenterfaceQT is launched without the KVM connected: 0 - 3%
After connecting the KVM: ~50%
After connecting the KVM to a video source: 40 - 60%
Connecting KVM first with video source, then opening OpenterfaceQT: 60 - 75%
VLC: 1.5%
OBS Studio: 30 - 40%

kevinzjpeng, 2025-03-04:
It seems that OpenterfaceQT is unable to directly utilize your GPU. To check if this is causing high CPU usage, you can try launching OpenterfaceQT with the following command. This may help reduce CPU usage by enabling GPU acceleration
```
QT_FFMPEG_DECODING_HW_DEVICE_TYPES=drm openterfaceQT
```
Let me know if it works, I will build this setting to auto enable the GPU in linux version.