TechxArtisanStudio/Openterface_QT #177: Serial port reset required multiples time per day

zorxd, 2025-02-20:
Hi

I am using application version 0.0.39.009.
Once in a while, keyboard and mouse stops working. I need to use the "reset serial port" function to get it working again.
Most of the times it works. But sometimes it's not enough. I then have to disconnect the USB cables and connect again.

kevinzjpeng, 2025-02-21:
Hi, can you share what is your host OS? Linux or windows? and there is a newer version, can you try.

https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/v0.1.0

zorxd, 2025-02-21:
I am using windows. I upgraded to version 0.1.0 and just got the same problem. Keyboard and mouse started working again after clicking "reset serial port".

kevinzjpeng, 2025-02-24:
Thanks for update. The serial issue could likely be due to an incorrect driver installation during the initial setup, or repeated plugging/unplugging of devices. Typically, using the "reset serial port" function helps clear the system status and resolve the problem. We’re actively working on adding clearer instructions and improving the software to address such problems more effectively.
Because your Opeterface is working now, I will close this issue.

closed 2025-02-24 by kevinzjpeng

zorxd, 2025-02-24:
Which driver should I use for it to work?

I think this issue should remain open as it is not resolved.

zorxd, 2025-02-27:
Hi
Still happening a lot. It looks like either a major software bug, or my hardware is defective. Please re-open until sorted out.