TechxArtisanStudio/Openterface_QT #184: Keyboard control work at the BIOS level

ajtalbot1, 2025-02-24:
This is also noted in your FAQ under "Why doesn't the keyboard control work at the BIOS level for some older target computers?"

Just got my Mini-KVM and plugged it into my old hanger queen computer. First step was to clear the BIOS setting but I was not able to access the BIOS with out plugin in a physical keyboard.

Motherboard: GA-970A-UD3 (https://www.gigabyte.com/Motherboard/GA-970A-UD3-rev-10-11)
BIOS: F8f (Latest)
CPU: AMD phenom II x2 555
16GB RAM
Openterface 0.1.1.052 (I tried the advanced, "Factory Reset HID Chip")

Keyboard does not at the BIOS level.  Works fine when a higher level OS driver takes over. 
Please let me know what I can do to help.
-Adam

kevinzjpeng, 2025-03-04:
Dear Adam,

Thank you for reaching out and providing detailed information about the issue you're experiencing.

We are aware of this BIOS-level keyboard compatibility issue with certain older motherboards, as noted in our FAQ: [Why doesn't the keyboard control work at the BIOS level for some older target computers? ](https://openterface.com/faq/#why-doesnt-the-keyboard-control-work-at-the-bios-level-for-some-older-target-computers). Unfortunately, there isn’t a definitive solution yet, as the root cause remains somewhat mysterious.

Since we don’t have access to older hardware like your GA-970A-UD3 motherboard, it’s challenging for us to test and troubleshoot this issue effectively. If you’re willing to assist, it would be incredibly helpful for us to improve our product.

Here’s how we can proceed:

1. I can implement a "keyboard mode switch" in the settings page of the Mini-KVM. This would allow you to toggle the device into a pure keyboard mode.
2. Once this feature is ready, I’ll let you know so you can test whether it resolves the BIOS-level keyboard functionality on your system.

If you’re open to helping, please let me know, and I’ll prioritize building and sharing this update with you. Your feedback will be invaluable in refining our product for users with similar setups.

ajtalbot1, 2025-03-04:
I am ready.  When you have the "keyboard mode switch" setup, please send me
the test version.  I am currently running the mini KVM client software on
my Windows system.
Thanks

kashalls, 2025-03-07:
@kevinzjpeng `lsusb -v -d 1a86:e329 | grep bInterfaceSubClass` returns

[image]

Which is apart of the HID on the serial convert. It needs to be set to all 1's for the bios to allow it.

kevinzjpeng, 2025-03-08:
@kashalls thanks for pointing out a way for me to test.

ajtalbot1, 2025-04-29:
Better late than never.  Tested against 0.3.9.117.  Works perfectly.
Thanks
Adam

closed 2025-04-29 by ajtalbot1