TechxArtisanStudio/Openterface_QT #197: Firmware license

sur5r, 2025-03-07:
With the firmware update functionality added in 2d37e9a8f20e8388cbdbb54ba1c95e3b274540dd, an embedded firmware images was added as well.

Given this is essentially a precombiled binary, AGPL doesn't work here.
So my question is: Which license applies here? Will the source for it be available at some point?

Reason I'm asking: I'm packaging Openterface-QT for Debian, see https://bugs.debian.org/1086052, so I need to know whether it will be able to stay in main or whether it needs to move to non-free.

Shipping the firmware as a separate, downloadable file would also help distribution-wise.

kashalls, 2025-03-07:
> Given this is essentially a precombiled binary, AGPL doesn't work here. So my question is: Which license applies here? Will the source for it be available at some point?

That's a good question. From what I understand is that the firmware is closed-sourced. Anyone can extract the firmware and make modification to it. If you look hard enough on the internet, you can find programs to modify it and that's ultimately how I helped the team add support for modifying parts of the firmware. 

I guess this relies on @youyoubilly having Redistribution Rights.

kevinzjpeng, 2025-03-08:
@sur5r You've brought up a valid point that I was previously unaware of. Indeed, you are correct that the firmware is licensed by Macro Silicon, despite it being in binary form. In next release, likely next week, I will separate the firmware from the code in order to properly align with the terms of the AGPL license. So you can hold on for your merge.

Finally, thanks for your hard work for the Debian packaging.

closed 2025-04-04 by kevinzjpeng