TechxArtisanStudio/Openterface_QT #210: App repeatedly scans for hidraw devices, causing stutter

justinas, 2025-03-24:
For whatever reason, my copy of Openterface_QT seems to be repeatedly executing the [getHIDDevice](https://github.com/TechxArtisanStudio/Openterface_QT/blob/807fb7c5a18f75ab851cb25171d902a01e4a5f96/video/videohid.cpp#L455) method once every 50 milliseconds or so. This seems to result to a very stuttery UI experience - everything, including mouse movements, video capture playback, and UI itself lags (updates only once every 1-2 seconds), as this scan is seemingly done in the UI thread.

It seems like this method is only called from this code:

https://github.com/TechxArtisanStudio/Openterface_QT/blob/807fb7c5a18f75ab851cb25171d902a01e4a5f96/video/videohid.cpp#L494-L504

which on the surface looks correct - after initially successfully opening the device, `hidFd` should get set to a positive value and further calls should not scan devices again. I'm not sure why that is not the case, perhaps there are some data races or undefined behavior at play.

I haven't been able to compile with debug symbols so far, so I didn't look much further into it.

<details>
<summary>
An excerpt from the console log:
</summary>

```
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.ui.mainwindow]: Set defualt camera to the Openterface camera...
[2025-03-24 15:16:37.768][Qt mainThread] [D][default]: Initializing audio...
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Audio Input Device name: "Family 17h/19h/1ah HD Audio Controller Headphones Stereo Microphone" , ID: "alsa_input.pci-0000_07_00.6.HiFi__Mic2__source"
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Audio Input Device name: "Family 17h/19h/1ah HD Audio Controller Digital Microphone" , ID: "alsa_input.pci-0000_07_00.6.HiFi__Mic1__source"
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Audio Input Device name: "Openterface Analog Stereo" , ID: "alsa_input.usb-MACROSILICON_Openterface_________-02.analog-stereo"
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Initializing audio with input device: ""
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Audio format details:
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Sample rate: 48000
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Channel count: 2
[2025-03-24 15:16:37.768][Qt mainThread] [D][opf.core.host.audio]: Sample size: 4
[2025-03-24 15:16:37.769][Qt mainThread] [D][opf.ui.camera]: Set Camera to videoOutput:  VideoPane(0x24650810) , device name:  "Openterface: Openterface"
[2025-03-24 15:16:37.769][Qt mainThread] [D][opf.ui.camera]: Camera connections set up
[2025-03-24 15:16:37.769][Qt mainThread] [D][opf.ui.camera]: Setting video output to:  ""
[2025-03-24 15:16:37.769][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:37.769][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:37.769][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:37.821][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:37.821][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:37.821][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:37.874][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:37.874][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:37.874][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:37.926][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:37.926][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:37.926][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:37.978][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:37.978][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:37.978][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:38.030][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:38.030][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:38.030][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:38.082][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:38.082][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:38.082][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:38.134][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:38.134][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:38.134][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[2025-03-24 15:16:38.186][Qt mainThread] [D][opf.ui.camera]: Input resolution:  std::pair(1920,1080)
[2025-03-24 15:16:38.186][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-03-24 15:16:38.186][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-03-24 15:16:38.186][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
```

</details>

If something like UB is truly at fault here, I wouldn't be surprised if this isn't easily reproducible by other people running e.g. official builds. I'm using NixOS. You can acquire this same version of Openterface_QT, compiled from source by having Nix installed, and running:

```
$ nix-build https://github.com/NixOS/nixpkgs/archive/1e5b653dff12029333a6546c11e108ede13052eb.tar.gz -A openterface-qt
$ ./result/bin/openterfaceQT
```

Doridian, 2025-03-24:
Can confirm exactly this is happening for me. I'm using a build from the `openterface_qt` AUR

kevinzjpeng, 2025-03-31:
I can confirm that this is indeed a flawed design, and I have just pushed a fix in [this commit](https://github.com/TechxArtisanStudio/Openterface_QT/commit/d70f6c3a4806d2388db93111bc6a3044183f6d3e#diff-e615b0c965ba727a5eaff46296641919392e377950bddcd54080517acd3980d2).

The root cause of the issue is our attempt to ensure that the software supports plug-and-play functionality. However, Qt does not natively support detection of USB serial device plug and unplug events. As a result, we had to implement a regular check of device information every second. Over time, as more features were added, the logic within this regular check became increasingly complex. The application continuously opened and closed the HID file descriptor, which led to a significant drop in keyboard and mouse performance, especially on Linux systems.

To address this, I have optimized the code to reduce the frequency of opening and closing the HID device. This change has significantly improved performance.

If you would like to try out this fix sooner, you can download the latest build from the GitHub Actions artifacts. This change will also be included in the next official release.

stop5, 2025-04-02:
I just tested the commit on my archlinux.
I can confirm that its now less checking the devices, but it seems to wait only 50 milliseconds:
```
[2025-04-03 00:13:33.685][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-04-03 00:13:33.685][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-04-03 00:13:33.685][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
[...]
[2025-04-03 00:13:33.737][Qt mainThread] [D][default]: Line:  ""
[2025-04-03 00:13:33.737][Qt mainThread] [D][default]: Line:  "DRIVER=hid-generic"
[2025-04-03 00:13:33.737][Qt mainThread] [D][default]: Line:  "HID_ID=0003:0000534D:00002109"
[2025-04-03 00:13:33.737][Qt mainThread] [D][default]: Line:  "HID_NAME=MACROSILICON Openterface"
```

when i enable the filelogging it creates an 1.1MiB file when i ran it for 18 seconds with an tmajority only printing out the current hid devices.

kevinzjpeng, 2025-04-04:
The issue is fixed in 0.3.1 release.

https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/0.3.1

closed 2025-04-04 by kevinzjpeng