TechxArtisanStudio/Openterface_QT #243: Poor handling of dead keys

zorxd, 2025-04-11:
Hi
I am trying to make a keyboard layout for the Canadian French keyboard.
I successfully mapped most keys. The problem is the use of dead keys. To write a character like "è", we press on the grave accent key and then the "e" key.

The problem is with openterface, I now need to press the grave accent key TWICE and then the "e" key for it to work.

Any example of another language working with dead keys? They are usually used for accented characters like êûîôïëç

0.3.1.194 windows 11

thanks

Chow411, 2025-04-16:
Thank you for reaching out and for your detailed explanation of the issue you’re experiencing with the Canadian French keyboard layout. We truly appreciate your feedback and your support for our software!

At the moment, we haven’t been able to reproduce this specific behavior with dead keys (where pressing the grave accent key twice is required for è). In our current implementation, we handle special characters like ê, û, î, ô, ï, ë, and ç using Unicode mapping—our debugging logs (QDebug) show the corresponding Unicode values, which we then map to the correct key inputs.

To help troubleshoot, could you please verify the following?

1. Keyboard Layout Consistency: Ensure that the keyboard layout matches across:

Your physical keyboard layout (target device)

Your host machine keyboard layout (Windows 11)

The layout selected in our software

Unicode vs. Dead Keys: If possible, try using Unicode mappings for accented characters to see if that resolves the issue.

If the problem persists, we’d be happy to assist further—feel free to share any additional details or logs that might help us investigate.

Thanks again for your patience and for helping us improve the software!

zorxd, 2025-04-16:
Hello
My physical keyboard, my host and my target keyboard layouts are all **French Canada** (not to be confused with French Canada Legacy or Canadian multilingual).
Here is a picture of my layout: https://kbdlayout.info/KBDCA/ (default iso105 arrangement)

There is no û key on my layout. é is the only accented character with a dedicated key. So I need to press the ^ dead key and then u to type a û. The bug is when using Openterface, I need to press the ^ key twice and then u to get a û.

I did try to use unicode for key mapping. Attached is my work in progress keyboard layout json file. Let me know if you have any comment.

[qwerty_frca.json](https://github.com/user-attachments/files/19777184/qwerty_frca.json)
Some rarely used characters such as £¢¤µ¶ are still not working properly, but my main issue is really the accented characters requiring a dead key. So my goal is to get the following one (and their capital variants) to work: êèëïüûùîôàâç.

thanks for your support

Chow411, 2025-06-11:
This seems to be related to a hardware issue where the CH9329 chip only supports the US keyboard layout best, all the keyboard layout just using the HID code match and sending now. We have no idea to fix this problem now.
Thank you your feedback and contribution.