TechxArtisanStudio/Openterface_QT #252: Deb 0.3.3 fails to detect hardware

youyoubilly, 2025-04-20:
Repost https://github.com/TechxArtisanStudio/Openterface/issues/20#issue-3006683786
created by @tunloop

> OS: Debian 12
> 
> I was previously using `0.0.46.032` and upgraded to 0.3.3 and now the hardware is undetected.
> 
> I've attempted disconnect and reconnect of the host and target usb-c cables, but it still says undetected. The stdout seems to indicate all the lower level stuff is found? Is there any way to get more detailed logs?
> 
> ```
> Start openterface...
> Original QT Media Backend: ""
> Current QT Media Backend: "ffmpeg"
> Set QT_QPA_PLATFORM to xcb
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
> [2025-04-19 15:17:39][139784553032064] {Debug}: Show window now
> [2025-04-19 15:17:39][139784553032064] {Debug}: Loading settings
> [2025-04-19 15:17:39][139784553032064] {Debug}: Enable log store now 
> [2025-04-19 15:17:39][139784553032064] {Debug}: Store log is  false
> [2025-04-19 15:17:39.427][139784553032064] [D][default]: Enable log store done
> [2025-04-19 15:17:39.427][139784553032064] [D][opf.host.layouts]: Loading keyboard layouts from directory: "/usr/bin/config/keyboards"
> [2025-04-19 15:17:39.427][139784553032064] [D][opf.host.layouts]: Found 0 layout files in filesystem.
> [2025-04-19 15:17:39.427][139784553032064] [D][opf.host.layouts]: Found 7 layout files in resources.
> Auto-check preference: enabled
> Checking if MS2109 is on Linux.
> Bus 007 Device 009: ID 534d:2109 MacroSilicon Openterface
> Checking if BRLTTY is installed.
> BRLTTY is not running. Good!
> Checking if driver is installed on Linux.
> Bus 007 Device 010: ID 1a86:7523 QinHeng Electronics CH340 serial converter
> Driver installation status: Installed (found via lsusb)
> Checking if MS2109 is on Linux.
> Bus 007 Device 009: ID 534d:2109 MacroSilicon Openterface
> Checking if BRLTTY is installed.
> BRLTTY is not running. Good!
> Checking if driver is installed on Linux.
> Bus 007 Device 010: ID 1a86:7523 QinHeng Electronics CH340 serial converter
> Driver installation status: Installed (found via lsusb)
> ```
> 
> [image]

Koustav-Samaddar, 2025-04-28:
I'm seeing the exact same behaviour in Ubuntu 24.02.2 w/ software version 0.3.3.

```Auto-check preference: enabled
Checking if MS2109 is on Linux.
Bus 005 Device 012: ID 534d:2109 MacroSilicon Openterface
Checking if BRLTTY is installed.
BRLTTY is running. It may interfere with device access.
Checking if driver is installed on Linux.
Bus 005 Device 013: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Driver installation status: Installed (found via lsusb)
Checking if MS2109 is on Linux.
Bus 005 Device 012: ID 534d:2109 MacroSilicon Openterface
Checking if BRLTTY is installed.
BRLTTY is running. It may interfere with device access.
Checking if driver is installed on Linux.
Bus 005 Device 013: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Driver installation status: Installed (found via lsusb)
[2025-04-27 19:56:44.187][128696447003840] [D][default]: Driver dialog rejected
```

slatkin, 2025-04-30:
I am seeing the same behaviour on Arch, with latest version 0.3.9.

junkeller, 2025-04-30:
To fix this issue I had to do following:
```
# Setup the hidraw permission
echo 'KERNEL== "hidraw*", SUBSYSTEM=="hidraw", MODE="0666"' | sudo tee /etc/udev/rules.d/51-openterface.rules 
sudo udevadm control --reload-rules
sudo udevadm trigger
```
Copied from https://github.com/TechxArtisanStudio/Openterface_QT/?tab=readme-ov-file#for-linux

But maybe you want to wait until there is a better solution:
https://github.com/TechxArtisanStudio/Openterface_QT/issues/203

closed 2025-06-11 by Chow411