TechxArtisanStudio/Openterface_QT #267: Fedora 42 fresh install segfault

FingerlessGlov3s, 2025-05-13:
Just did a fresh install of Fedora 42, and then install OpenterfaceQT RPM. When I try open it, it errors with a segfault, without the openterface device connected.
System is fully up to date and have rebooted since installing the RPM.

```
~$ openterfaceQT 
Auto-check preference: enabled
Checking if MS2109 is on Linux.
libusb initialized successfully.
MS2109 not exist, so no Openterface plugged in
CH341 not exist, so no Openterface plugged in
Segmentation fault (core dumped)
~$ uname -a
Linux fedora 6.14.5-300.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Fri May  2 14:16:46 UTC 2025 x86_64 GNU/Linux
```

Opening it with the device connected, it get the prompt to add the udev rules. After copying n pasting and running the commands, it opens and then sound from the app screeches at me, until I unplugged the openterface. I saw in the stdout
```
[2025-05-13 13:29:54.688][Qt mainThread] [D][opf.host.hid]: Retrying sendFeatureReportLinux...
[2025-05-13 13:29:54.688][Qt mainThread] [D][opf.host.hid]: Failed to open HID device ( "/dev/hidraw3" ). Error: Permission denied
```

I have to run the process as root, for it to work. Also tried a manual reload of udevadm, and replugging the USB, but always need to be root.

Chow411, 2025-05-27:
At version 0.3.13, we have fixed this issue by updating the hidraw udev rule. Please give it a try.

Chow411, 2025-05-28:
Yes, it misjudged that it didn't get the latest firmware due to network issues and some logical issues with the server. We will fix this problem at next version.

Chow411, 2025-06-11:
The latest version has updated the firmware update logic and URL, we close this issue.

closed 2025-06-11 by Chow411