TechxArtisanStudio/Openterface_QT #269: Video not refreshing

CLACyber, 2025-05-20:
When connecting the kvm I see what is initially on the screen but the image will never refresh, I know the keyboard and mouse are working because when I unplug and replug in the device I can see what was typed but then the image will stay static. This is for a windows 10 laptop. The device itself was tested on different computer connecting to the same machine and it works fine on everything else except this laptop. Has anyone else experienced the same? Software and firmware were updated to the latest.

Chow411, 2025-05-22:
This issue might be related to the connection between the host and KVM endpoints or you could try using a shorter USB cable as a workaround. If you purchased an official kit, please switch to the black cable instead and give a try.

CLACyber, 2025-05-22:
still seems to be the same problem using the original cables, the only difference is that I'm using a display port to HDMI converter. This setup I've tested with other host machines to the same target machine but only this laptop doesnt seem to work with getting the video signal to refresh on its own. For the laptop host machine I do see that this laptop has 2 video adapters a nvidia one and intel one, disabling one or the other doesnt seem to make a difference.

Chow411, 2025-05-23:
Our device can be recognized in Windows and we can use the built-in camera software in Win10 to check if it can acquire the frame reflash. We should also verify that the camera name is indeed "openterface" in the Device Manager.
And please enable all log settings by navigating to preference > Genral > Enable file logging and allow all the log Core Serial User Interface Host. This will allow us to access and review all logs recorded in a text file for further investigation.

CLACyber, 2025-05-23:
Turned out to be a realtek driver issue. works fine now!

closed 2025-05-23 by CLACyber