TechxArtisanStudio/Openterface_QT #275: Linux Mint Experiance with Openterface

BertQuodge, 2025-05-25:
Hi

I tried to use my Openterface today on a Linux Mint 21.3 Cinnamon today & I thought I'd share my experience. I decided to test the Openterface on Linux Mint in preparation for using the unit on another Linux system at a work site.

I initially downloaded the Debian installation file and when I tried to install I received a "dependency not satisfied libqt6core6 (>=6.4.2)" error. Without knowing the impact to the rest of my PC I didn't want to update libqt6core6, just in case.

I then tried the flatpak. If the openterface is not connected to the PC and the flatpack is started from the start menu nothing happens. If I start the flatpak from the cli I see a message about the openterface not being connected. Why doesn't the GUI say also this?

After connecting the openterface and starting the flatpack I received a message about adding udev rules, which I then added & rebooted. The flatpack now tells me, on every time I start it, that I need to update the firmware on the openterface and that a driver has been installed and I need a system restart for the changes to take effect. If I click yes to restart my PC the openterface flatpak just starts up normally. I manually rebooted the PC. On starting the flatpack software I'm again prompted that I need to update the firmware on the openterface and that a driver has been installed and I need a system restart for the changes to take effect. I can also see errors on the cli, that started the flatpack, that says "Failed to load module "xapp-gtk3-module" & "Qt: Session management error: could not open network socket".

Once I enter the flatpack software I tried to update the openterface firmware, as requested. If I go to advance/firmware update I get told my current firmware version is a string of 0s and the latest firmware version. lsusb shows that the openterface is connected. On clicking yes to update the firmware I received a message saying "An error occurred during the firmware update". The cli has a message saying "Failed to open HID device /dev/hidraw0 Error Permission Denied.

I then ran the flatpack as root (which isn't a good idea) and I get prompted about needing a system restart for the driver. After this I tried to update the openterface firmware. The flatpack can now see the openterface's firmware version and it allows me to update the firmware. The flatpack software exited and I removed & reinstalled the openterface to the PC's USB port.

I tried to run the flatpack again and after another "system restart required for the driver" message the openterface firmware version is again showing lots of 0s. I started the flatpack as a root user and I can now use the openterface. 

FYI, the environment helper is all green ticks except the latest firmware, which shows my version is many 0s, if I run as a normal user. If I run as root the environment helper doesn't display.

Given the issues I've had I'm going to take a keyboard and mouse with me to site instead of the openterface, which is a shame.

Chow411, 2025-05-27:
Thanks your feedback:
Our flatpak version was updated over the weekend, and the latest version should be 0.3.13. We have updated the udev rules to address hidraw permission issues by using the access method to ensure users can access the device. If the firmware needs to be updated even though the latest version is still present, it may be due to network connectivity issues. If lastest version displays as 00000, it indicates a network issue. To confirm proper functionality, you can disable automatic checks in Advance -> Environment Checking. Since we have updated to version 0.3.13, please try it out and provide feedback.

BertQuodge, 2025-05-28:
> Thank you for sharing your detailed experience — it's extremely helpful!
> 
>     * We’ve recently released an update to the Flatpak version that should address some of the issues you encountered. Specifically, we've added proper udev rules and improved access permissions for /dev/hidraw
> 
>     * Regarding the firmware update issue: We're aware that some users experience difficulties when checking or updating firmware. This is due to occasional unavailability of our firmware update server, combined with a bug in the current software version where network timeouts are incorrectly interpreted as the firmware being up-to-date. We sincerely apologize for this inconvenience, and we’re working to fix this in the next release to ensure more reliable and accurate firmware checks.

Thanks for your help.

My first attempt at installing the flatpack was on Sunday 25 May at 1600 UTC, so I might have the latest version.

I tried to run 'flatpak upgrade com.openterface.openterfaceQT' but my system says the flatpack is up to date. I then tried a 'flatpak remove com.openterface.openterfaceQT' followed by a 'flatpak install com.openterface.openterfaceQT'. I then deleted the '/etc/udev/rules.d/51-openterface.rule' file. On starting the openterface flatpack, as a user, I'm prompted to add the udev rules, which I do (as root). I then restart the PC and when I add the openterface hardware I can see only "root:root" has access to /dev/hidraw0, which also has 600 permissions. I can't say for sure, but the udev rules I now have look the same as I had before. As I completed previously, if I add my "bodge" udev rule below, '/dev/hidraw0' gains 666 permissions, meaning the flatpack can access the his device when run as a user.

My "bodge" udev rule:
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="534d", ATTRS{idProduct}=="2109", MODE="0666"

I think I have the latest flatpak, but the udev rules are not working for me.

FYI, on my system, when I add the Openterface, lsusb reports these devices are added:
Bus 003 Device 025: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 003 Device 024: ID 534d:2109 MacroSilicon Openterface
Bus 003 Device 023: ID 1a40:0101 Terminus Technology Inc. Hub 

Thanks