TechxArtisanStudio/Openterface_QT #278: Advance -> Firmware Update does nothing on latest Windows Release

W3AXL, 2025-06-02:
Clicking the `Advance` -> `Firmware Update...` button does not appear to do anything in the latest Windows release.

[image]

Checking the environment shows that I need new firmware, but there currently appears to be no way to upgrade it.

[image]

dataneim, 2025-06-04:
same issue here

tayvermtos, 2025-06-05:
Same issue here. But on my end it just shuts down the Windows App and I cannot seem to start it again. In task manager I cannot see the app running either, so it must be some form background task which I have no idea how to look for. 
Please advise.
Thanks

Chow411, 2025-06-06:
We have investigated the issue and determined that it is caused by a bug in the firmware update logic. We plan to address this problem in next release.

Chow411, 2025-06-09:
We had fix this bug please give it a try. [https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/0.3.15](url)

dataneim, 2025-06-09:
PROBLEM: Version 0.3.15  openterfaceQT-portable.exe does not even open in windows 11 24H2  now !! the moment double click the file, as if never run it.

tayvermtos, 2025-06-11:
Yup confirmed latest release 0.3.16 fixes the issue and I managed to upgrade the firmware without any issues!
Thanks!

dataneim, 2025-06-11:
I just downloaded portable  0.3.16 , still does not even open in windows 11 24H2 , the moment double click the file, as if never run it.

I tried from another win 11 computer, same issue. Can it be because of portable and setup files are different ??

lloydys15, 2025-06-11:
> I just downloaded portable 0.3.16 , still does not even open in windows 11 24H2 , the moment double click the file, as if never run it.
> 
> I tried from another win 11 computer, same issue. Can it be because of portable and setup files are different ??

i had the install version working but i just downloaded the portable aswell and thats working for me too, maybe try right click run as admin ?

dataneim, 2025-06-11:
i tried run as admin, same issue. i also played with other settings like compatibility mode for win 8 , win 7, disable fullscreen optimization, use lagacy display., all does not work.

currently only the portable version 0.3.12.129 is working with known issues of course, anything above does not work on my two Win 11 PCs

Chow411, 2025-06-11:
Or try installing using an installer and then running it once before uninstalling it, perhaps some registry cache in Win11 is causing it to not run properly. @dataneim