TechxArtisanStudio/Openterface_QT #283: Application crash on Windows for version 3.14

John3791, 2025-06-05:
When running the Windows portable version 0.3.14 the application would crash immediately. Here is the information from the Windows Application log:

```
Log Name:      Application
Source:        Application Error
Date:          6/5/2025 12:42:24 PM
Event ID:      1000
Task Category: Application Crashing Events
Level:         Error
Keywords:      
Description:
Faulting application name: openterfaceQT-portable.exe, version: 0.0.0.0, time stamp: 0x68415949
Faulting module name: openterfaceQT-portable.exe, version: 0.0.0.0, time stamp: 0x68415949
Exception code: 0xc0000005
Fault offset: 0x00000000006d9aeb
Faulting process id: 0x0x5C84
Faulting application start time: 0x0x1DBD651F5E1AA8F
Faulting application path: C:\Programs\Openterface\openterfaceQT-portable.exe
Faulting module path: C:\Programs\Openterface\openterfaceQT-portable.exe
Report Id: 2f209d93-94a3-45e4-8d62-a2ad453f003d
Faulting package full name: 
Faulting package-relative application ID: 
```

The application works correctly when I run version 0.3.13.

iball, 2025-06-06:
Same thing happens with latest 3.15 version.

Faulting application name: openterfaceQT.exe, version: 0.0.0.0, time stamp: 0x6842a252
Faulting module name: openterfaceQT.exe, version: 0.0.0.0, time stamp: 0x6842a252
Exception code: 0xc0000005
Fault offset: 0x00000000006d9d7b
Faulting process id: 0x17EA8
Faulting application start time: 0x1DBD6FFB2D7988E
Faulting application path: C:\Program Files (x86)\Openterface Mini-KVM\openterfaceQT.exe
Faulting module path: C:\Program Files (x86)\Openterface Mini-KVM\openterfaceQT.exe
Report Id: 78e32850-6976-4d68-ace2-a727bf26299d
Faulting package full name: 
Faulting package-relative application ID:

lloydys15, 2025-06-10:
windows 11 24h2 fully patched.

lloydys15, 2025-06-10:
issue appears to be resolved in latest build 0.3.16.131 thanks @Chow411 !!

Chow411, 2025-06-11:
The issue solved in [lastest release](https://github.com/TechxArtisanStudio/Openterface_QT/releases/tag/0.3.16)

closed 2025-06-11 by Chow411