TechxArtisanStudio/Openterface_QT #300: Portable version v0.3.17 does not work but installer v0.3.17  works.

dataneim, 2025-06-29:
I have tried on two different Windows 11 PCs, both results the same problem with portable v0.3.17, but not with the installer v0.3.17 .

Portable version does not even open up when you double click the exe file. No window pop up, no error message whatsoever.

Then I installed the setup v0.3.17 on both PCs, both installed versions run, giving a starting pop up message that the Driver is not installed (see screenshot). Then I installed the driver CH341SER.EXE (USB to serial port Windows driver and DLL library), working nicely.

[image]

I think the problem with portable version is that at start up if it cannot see any driver, it cannot even start the window, hence user cannot see any opened program no matter how many times he double click the file. 

Is there a way that you include these drivers inside the portable version, or at least you allow a ZIP file that the portable version can be extracted into a folder together with driver dll files, so that portable exe file can seek the drivers within the same folder ? 

My problem is, I want to use openterface in my work laptop, I am allowed to use portable version but not allowed to install from setup file due to admin policy. That's why using portable version is critical for me, and perhaps for many others here.

I see the driver CH341SER.EXE  also has ZIPPED file version where all the driver files are inside. 
https://www.wch.cn/downloads/CH341SER.EXE.html

Pyromane, 2025-07-10:
Older portable versions work without any problems, only the latest ones no longer do.

Pyromane, 2025-07-26:
Portable Version v0.3.19 cannot be started either.

Chow411, 2025-07-31:
The driver can be access from url: https://www.wch.cn/downloads/CH341SER_EXE.html

Pyromane, 2025-08-03:
v0.3.12 openterfaceQT-portable.exe works without any problems

v0.3.13 openterfaceQT-portable.exe does not start
v0.3.14 openterfaceQT-portable.exe does not start
v0.3.16 openterfaceQT-portable.exe does not start
v0.3.17 openterfaceQT-portable.exe does not start
v0.3.18 openterfaceQT-portable.exe does not start
v0.3.19 openterfaceQT-portable.exe does not start

> The driver can be access from url: https://www.wch.cn/downloads/CH341SER_EXE.html

Unfortunately, I cannot see any connection with the problem described here.

Chow411, 2025-09-15:
We think it is maybe the protable openssl libs problem, could you install the openssl at https://slproweb.com/products/Win32OpenSSL.html

> v0.3.12 openterfaceQT-portable.exe works without any problems
> 
> The tragedy continues: v0.3.13 openterfaceQT-portable.exe does not start v0.3.14 openterfaceQT-portable.exe does not start v0.3.16 openterfaceQT-portable.exe does not start v0.3.17 openterfaceQT-portable.exe does not start v0.3.18 openterfaceQT-portable.exe does not start v0.3.19 openterfaceQT-portable.exe does not start v0.4.3 openterfaceQT-portable.exe does not start
> 
> [@Chow411](https://github.com/Chow411) [@kevinzjpeng](https://github.com/kevinzjpeng) Can we help you troubleshoot the problem in any way?

dataneim, 2025-09-16:
installed openssl in windows 11, reset the PC,   portable ver 0.4.3 and ver 0.3.19 still problematic.

Pyromane, 2026-09-18:
In my opinion, this issue can be closed because the bug no longer exists in the current versions.