TechxArtisanStudio/Openterface_QT #307: deb arm64 package contains x86-64 binary

rondie, 2025-07-17:
the latest package, and older ones from what I've tried, for arm64 contain the x86-64 binary

on a debian system:
wget https://github.com/TechxArtisanStudio/Openterface_QT/releases/download/0.3.18/openterfaceQT.linux.arm64.deb
dpkg -x openterfaceQT.linux.arm64.deb .
file ./usr/bin/openterfaceQT

./usr/bin/openterfaceQT: ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=ba02aea64ecb8c9c494d1a026900b80585f665f2, for GNU/Linux 3.2.0, not stripped

rondie, 2025-07-17:
putting the [output logs](https://productionresultssa15.blob.core.windows.net/actions-results/4444d7db-de87-4d0a-a71c-e1a588d5194b/workflow-job-run-21a54c2f-dbf2-5b66-b5e9-300f11390106/logs/job/job-logs.txt?rsct=text%2Fplain&se=2025-07-17T11%3A43%3A09Z&sig=wKdmu%2FZAaADTAsOXzxze83mBu4ZhDrl1xnsKdlhHRlo%3D&ske=2025-07-17T19%3A54%3A07Z&skoid=ca7593d4-ee42-46cd-af88-8b886a2f84eb&sks=b&skt=2025-07-17T07%3A54%3A07Z&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skv=2025-05-05&sp=r&spr=https&sr=b&st=2025-07-17T11%3A33%3A04Z&sv=2025-05-05) through claude.ai (don't hate me) 
it seems the used 22.04 runner is x86-64 instead of arm64.

tward800, 2025-07-20:
Came here to post the same thing.  Glad to see it's already reported.

`~/Downloads/check/usr/bin$ file openterfaceQT 
openterfaceQT: ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=ba02aea64ecb8c9c494d1a026900b80585f665f2, for GNU/Linux 3.2.0, not stripped
`

tward800, 2025-07-20:
I also tried older releases back to 3.13 and even extracted the 3.18.linux.arm RPM.  They all show x86-64 in file output.

tayken, 2025-07-27:
I tried v0.0.6, v0.1.0, and v0.1.1, they all had 
```bash
$ dpkg-deb -R openterfaceQT.linux.arm64.deb openterface
$ file openterface/usr/bin/openterfaceQT 
openterface/usr/bin/openterfaceQT: ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=9cf70c53e9cc943d215e0a4be8b42de1dbb83cae, for GNU/Linux 3.2.0, not stripped
```
with `BuildID[sha1]` value being different between versions. I think arm64 build never worked as I see that we're downloading the amd64 version of Qt packages in [build logs](https://github.com/TechxArtisanStudio/Openterface_QT/actions/runs/16492628734/job/46630796674#step:6:67).

I think the issue is not using x86-64 based image but using the `docker/setup-qemu-action` wrong. I only saw it being used for building different docker image files. Seems like there are 2 paths forward:
1. Use and arm64 runner to build the arm64 .deb and .rpm package files. Arm images are listed here: https://github.com/actions/partner-runner-images
2. Cross compile on x86-64 for arm64. There are tutorials online for this but here is a [Qt forum post](https://forum.qt.io/topic/161305/using-cross-compilation-for-arm64-in-the-ubuntu-24-04-x86-environment) which can be helpful.