TechxArtisanStudio/Openterface_QT #313: openterfaceQT doesn't exit cleanly on Linux (segfault)

tayken, 2025-07-27:
Whether there is an Openterface KVM connected to the system or not, the openterfaceQT application doesn't exit cleanly. Instead it always exits with `Segmentation fault (core dumped)` message.

* With device connected:
```bash
[2025-07-27 20:01:06.005][137576012844224] [D][opf.ui.mainwindow]: MainWindow destructor called
[2025-07-27 20:01:06.005][137576012844224] [D][default]: Stop camera data...
[2025-07-27 20:01:06.006][137576012844224] [D][default]: Camera data stopped.
[2025-07-27 20:01:06.006][137576012844224] [D][opf.core.host.audio]: Disconnecting audio thread.
[2025-07-27 20:01:06.006][137576012844224] [D][default]: Audio manager stopped.
[2025-07-27 20:01:06.006][137576012844224] [D][default]: Stopping camera..
[2025-07-27 20:01:06.006][137576012844224] [D][opf.device.hid]: Stopping VideoHid timer.
[2025-07-27 20:01:06.006][137576012844224] [D][default]: Camera is already stopped
[2025-07-27 20:01:06.006][137576012844224] [D][opf.core.serial]: Close serial port
[2025-07-27 20:01:06.316][137576012844224] [D][default]: Camera stopped.
[2025-07-27 20:01:06.316][137576012844224] [D][default]: Stopping camera..
[2025-07-27 20:01:06.316][137576012844224] [D][opf.device.hid]: Stopping VideoHid timer.
[2025-07-27 20:01:06.317][137576012844224] [D][default]: Camera is already stopped
[2025-07-27 20:01:06.317][137576012844224] [W][default]: failed to stop capture
[2025-07-27 20:01:06.317][137576012844224] [D][opf.ui.mainwindow]: MainWindow destroyed successfully
[2025-07-27 20:01:06.317][137576012844224] [D][opf.core.host.audio]: Disconnecting audio thread.
[2025-07-27 20:01:06.363][137576012844224] [D][opf.core.serial]: Destroy serial port manager.
[2025-07-27 20:01:06.363][137576012844224] [D][opf.core.serial]: Disconnecting SerialPortManager from hotplug monitor
[2025-07-27 20:01:06.363][137576012844224] [D][opf.core.serial]: SerialPortManager disconnected from hotplug monitor
[2025-07-27 20:01:06.364][137576012844224] [D][opf.core.serial]: Close serial port
[2025-07-27 20:01:06.364][137576012844224] [D][opf.core.serial]: Serial port is not opened.
Segmentation fault (core dumped)
```
* Without device connected
```bash
[2025-07-27 20:04:48.777][136177201044672] [D][opf.ui.mainwindow]: MainWindow destructor called
[2025-07-27 20:04:48.778][136177201044672] [D][default]: Stop camera data...
[2025-07-27 20:04:48.778][136177201044672] [D][default]: Camera data stopped.
[2025-07-27 20:04:48.778][136177201044672] [D][opf.core.host.audio]: Disconnecting audio thread.
[2025-07-27 20:04:48.778][136177201044672] [D][default]: Audio manager stopped.
[2025-07-27 20:04:48.778][136177201044672] [D][default]: Stopping camera..
[2025-07-27 20:04:48.778][136177201044672] [D][opf.device.hid]: Stopping VideoHid timer.
[2025-07-27 20:04:48.778][136177201044672] [W][opf.ui.camera]: Camera is null, cannot stop
[2025-07-27 20:04:48.778][136177201044672] [D][opf.core.serial]: Close serial port
[2025-07-27 20:04:48.778][136177201044672] [D][opf.core.serial]: Serial port is not opened.
[2025-07-27 20:04:49.081][136177201044672] [D][default]: Camera stopped.
[2025-07-27 20:04:49.081][136177201044672] [D][default]: Stopping camera..
[2025-07-27 20:04:49.081][136177201044672] [D][opf.device.hid]: Stopping VideoHid timer.
[2025-07-27 20:04:49.081][136177201044672] [W][opf.ui.camera]: Camera is null, cannot stop
[2025-07-27 20:04:49.081][136177201044672] [D][opf.ui.mainwindow]: MainWindow destroyed successfully
[2025-07-27 20:04:49.081][136177201044672] [D][opf.core.host.audio]: Disconnecting audio thread.
[2025-07-27 20:04:49.115][136177201044672] [D][opf.core.serial]: Destroy serial port manager.
[2025-07-27 20:04:49.115][136177201044672] [D][opf.core.serial]: Disconnecting SerialPortManager from hotplug monitor
[2025-07-27 20:04:49.115][136177201044672] [D][opf.core.serial]: SerialPortManager disconnected from hotplug monitor
[2025-07-27 20:04:49.116][136177201044672] [D][opf.core.serial]: Close serial port
[2025-07-27 20:04:49.116][136177201044672] [D][opf.core.serial]: Serial port is not opened.
Segmentation fault (core dumped)
```

It looks like it's trying to stop things twice and accessing something it shouldn't access.

tayken, 2025-07-27:
Seems like in both cases these messages are repeated twice:
```bash
[D][opf.core.host.audio]: Disconnecting audio thread.
[D][default]: Stopping camera..
[D][opf.device.hid]: Stopping VideoHid timer.
[D][opf.core.serial]: Close serial port 
```

* I don't think debug log message ordering fully helps with timing but it's helpful as we can probably see what's being repeated. 
* `Disconnecting audio thread.` message is called part of `void AudioManager::disconnect()` function. This function is called both in `AudioManager::~AudioManager()` and in `void MainWindow::stop()` functions. 
* `Stopping camera..` message is part of `void CameraManager::stopCamera()` function. This function is called in `void MainWindow::stop()` and `MainWindow::~MainWindow()` functions.
*  `Stopping VideoHid timer.` message is part of `void VideoHid::stop()` function. This function is called in `void CameraManager::stopCamera()`, which in turn is called in `void MainWindow::stop()` and `MainWindow::~MainWindow()` functions.
* `Close serial port` message is part of `void SerialPortManager::closePort()`. This function is called in `void SerialPortManager::stop()`, `SerialPortManager::~SerialPortManager()` and `void MainWindow::stop()` functions.
* `void MainWindow::stop()` function is also called in `MainWindow::~MainWindow()` function.
* Checking out the crash log shows that the problematic line is in `void SerialPortManager::closePort()` function and is https://github.com/TechxArtisanStudio/Openterface_QT/blob/d58f3cf6c4edb80bddbcd90b06e2cfd5832a2384/serial/SerialPortManager.cpp#L631

It looks like we're calling things multiple times but only serial port is messing things up. If I comment out the problematic line, compile and run openterfaceQT without any device connected, I get
```bash
[2025-07-28 01:34:51.829][128739812973760] [D][opf.ui.mainwindow]: MainWindow destructor called
[2025-07-28 01:34:51.829][128739812973760] [D][default]: Stop camera data...
[2025-07-28 01:34:51.829][128739812973760] [D][default]: Camera data stopped.
[2025-07-28 01:34:51.830][128739812973760] [D][opf.core.host.audio]: Disconnecting audio thread.
[2025-07-28 01:34:51.830][128739812973760] [D][default]: Audio manager stopped.
[2025-07-28 01:34:51.830][128739812973760] [D][default]: Stopping camera..
[2025-07-28 01:34:51.830][128739812973760] [D][opf.device.hid]: Stopping VideoHid timer.
[2025-07-28 01:34:51.830][128739812973760] [W][opf.ui.camera]: Camera is null, cannot stop
[2025-07-28 01:34:51.830][128739812973760] [D][opf.core.serial]: Close serial port
[2025-07-28 01:34:51.830][128739812973760] [D][opf.core.serial]: Serial port is not opened.
[2025-07-28 01:34:52.130][128739812973760] [D][default]: Camera stopped.
[2025-07-28 01:34:52.130][128739812973760] [D][default]: Stopping camera..
[2025-07-28 01:34:52.130][128739812973760] [D][opf.device.hid]: Stopping VideoHid timer.
[2025-07-28 01:34:52.130][128739812973760] [W][opf.ui.camera]: Camera is null, cannot stop
[2025-07-28 01:34:52.131][128739812973760] [D][opf.ui.mainwindow]: MainWindow destroyed successfully
[2025-07-28 01:34:52.131][128739812973760] [D][opf.core.host.audio]: Disconnecting audio thread.
[2025-07-28 01:34:52.171][128739812973760] [D][opf.core.serial]: Destroy serial port manager.
[2025-07-28 01:34:52.171][128739812973760] [D][opf.core.serial]: Disconnecting SerialPortManager from hotplug monitor
[2025-07-28 01:34:52.171][128739812973760] [D][opf.core.serial]: SerialPortManager disconnected from hotplug monitor
[2025-07-28 01:34:52.171][128739812973760] [D][opf.core.serial]: Close serial port
[2025-07-28 01:34:52.171][128739812973760] [D][opf.core.serial]: Serial port is not opened.
[2025-07-28 01:34:52.474][128739812973760] [D][opf.device.manager]: Stopping hotplug monitoring
[2025-07-28 01:34:52.474][128739812973760] [I][opf.device.hotplug]: Hotplug monitoring stopped
[2025-07-28 01:34:52.474][128739812973760] [D][opf.device.hotplug]: All callbacks cleared
[2025-07-28 01:34:52.475][128739812973760] [D][opf.device.hotplug]: Hotplug monitor destroyed
```

Chow411, 2025-08-11:
We‘ve debug the program. During program shutdown, we found that the ~MainWindow destructor calls SerialPortManager::getInstance().stop() multiple times. However, the eventCallback in the SerialPortManager still points to memory that has already been freed, causing a crash. We have updated the implementation of stop() to address this issue. This fix will be included in the next version. And thank you for your notice.

closed 2026-03-05 by Chow411